jsamr/bootiso #15: Add GPT option

Sporif, 2018-07-12:
I think a GPT option is needed for Windows 10 ISOs.  Currently mbr+rsync is used, but this causes some kind of weird missing driver error in the installation setup. I tried [windows2usb](https://github.com/ValdikSS/windows2usb) with mbr and it showed the same issue, but with gpt it worked.

jsamr, 2018-07-12:
@Sporif Thanks for the report, I'll check this up.

jsamr, 2018-07-14:
@Sporif I have reports of other users for which Windows10 boot worked great. Can you provide more details on the steps you went through ? Would you have screenshots or error codes Windows10 gave ? Or any reliable source confirming this issue with MBR + UEFI?

Thanks a lot for your help.

jsamr, 2018-07-14:
@Sporif
Thanks. I will add a feature request for GPT and wait to see if it gets upvoated. There is some work to do and I'd rather support more important features first. But if it's really causing driver issues in W10 I might reconsider my priorities.

So we need to isolate potential causes here. I suggest :
- checksum the ISO
- try bootiso with `--mrsync -t ntfs` options on the same ISO

Sporif, 2018-07-14:
Well this is quite strange, the issue seems to have fixed itself. Did the exact same actions and no error popped up.

jsamr, 2018-07-14:
You can confirm not using `-t ntfs` ?

Sporif, 2018-07-14:
Yep, did it with only path like before. So mbr+fat32+rsync was used.

jsamr, 2018-07-14:
@Sporif OK I'll close then.

HassanMirza01, 2018-07-25:
This issue came due to MBR booting on a GPT Partitioned HDD, or UEFI selected bios, we can't do it like vice virsa, GPT burned stuff should applied on GPT and mbr on mbr style....

HassanMirza01, 2018-07-25:
as i told before, for me any Linux distro is booting fine with both GPT and MBR, both with DD and rsync methods, Linux distros dont give any issue, Windows ISO are pain in ass, they need special treatment, you can never ever boot GPT booted on MBR scheme and MBR booted on GPT scheme atleast for windows, i have experience of it many times, all my issues solved now as i can make gpt too now, adding in my script, removing all syslinux stuff too, as i got another easy idea of getting it with grub install, ;)

jsamr, 2019-01-30:
I am considering a `--gpt` flag in some upcoming release.

Seen repro [here](https://www.linuxuprising.com/2018/07/bootiso-easy-iso-to-bootable-usb-drive.html): 
[image]

HassanMirza01, 2019-01-30:
i have it working fine here in my script... i will do a merge request for you soon ;)

jsamr, 2019-01-30:
@HassanMirza01 Thanks! You can propose PRs on branch `next`

MichaelFirsov, 2019-10-22:
Confirm - that's the really strange issue and as far as I know it has nothing to do with MBR/UEFI and etc. Here are my tests:

1) In _Windows 7 USB DVD Download tool_ I create a bootable usb (**Transcend usb3 8GB**) using the en_windows_10_consumer_editions_version_1909_x64_dvd_be09950e.iso - error

2) In _Windows 7 USB DVD Download tool_ I create a bootable usb (**Transcend usb3 8GB**) using the en_windows_10_business_edition_version_1809_updated_sept_2018_x64_dvd_f0b7dc68.iso - error

3) In _Windows 7 USB DVD Download tool_ I create a bootable usb (**Transcend usb3 8GB**) using the ru_windows_10_business_edition_version_1809_updated_sept_2018_x86_dvd_e75f8430.iso - error

4)  In _Windows 7 USB DVD Download tool_ I create a bootable usb (**Transcend usb3 8GB**) using the en_**windows_7**_professional_with_sp1_x64_dvd_u_676939.iso - success

--------------------------------------------------------------------------------------------------
5) - 6) ... In _Windows 7 USB DVD Download tool_ I create a bootable usb using any of the aforementioned images onto ANOTHER **Transcend usb3 8GB** stick - exactly the same as the first one (these two usb sticks have been purchased at the same time) - works perfect!

jsamr, 2019-10-31:
@MichaelFirsov Thank you for this thorough report. It's very hard to diagnose such heisenbugs! I have a feeling it might be related to #48, in a different way. There might be a race condition somewhere that involves i/o operations with a tool used by bootiso. Because it is hard to diagnose, I need to wait for more feedback. Your was definitely helpful, thanks.

MichaelFirsov, 2019-10-31:
jsamt, thank you for your reply!

jsamr, 2020-04-29:
I believe now it was related to #48. Fixed in the next branch.
4.0.0 release is coming out soon.

closed 2020-04-29 by jsamr

jsamr, 2020-05-09:
Also, GPT will be default partition table scheme when UEFI support is detected in mount+rsync mode.

jsamr, 2020-05-11:
I could reproduce the failure with v3 which was not cause by GPT, but a `sync` call missed in background. I am now certain this bug and #48 are related, and I can confirm this issue has been fixed in the upcoming v4!