jsamr/bootiso #16: Automatically detect shaX and md5 files to check sums against ISO

jsamr, 2018-07-12:


SibrenVasse, 2019-01-30:
I've made an implementation which looks in the directory of the ISO for the files:
md5sums*
MD5SUMS*
sha1sums*
SHA1SUMS*
sha256sums*
SHA256sums*
$filename.md5
$filename.sha1
$filename.sha256
$filename.sha512

Is that everything?

SibrenVasse, 2019-01-31:
I also wanted to support hashfiles provided by distros. Debian does it like this:
https://cdimage.debian.org/debian-cd/current/amd64/iso-cd/
and arch like this:
https://mirrors.evowise.com/archlinux/iso/2019.01.01/
Or do you only want to support the $filename.$hash files?

Thus far, I've got these flags:
--no-hash-check) disable hash checking altogether
--force-hash-check) fail if no valid hash is found

Maybe --hash-file <filename> would be a nice flag name for the extra file to check?

SibrenVasse, 2019-01-31:
Well the hash files contain the filenames along with the hashes. So we can check if a md5sums* file is relevant to the iso, and only fail if the filename is in there.

So as I've currently implemented it:

### Flow I : $filename.$hash exists
1. if --no-hash-check is not set, automatically check the sum
2. if the test fails:
    a. if `--force-hash-check`, failAndExit
    b. else print an echowarn message and prompt the user

### Flow II : hash candidate exists (md5sums*)
1. if `--no-hash-check` is not set and md5sums* contains filename  
2. Do I.2

### Flow III, --hash-file set
1. If hash file does not exists, failAndExit
2. Do I.2 for all hash types

### Flow IV, `--force-hash-check` set
1. After querying all files, no applicable hash is found, failAndExit

SibrenVasse, 2019-01-31:
Also about the dependencies:
{"md5sum", "sha1sum", "sha256sum", "sha512sum"} reside in coreutils on debian and archlinux.
Do we need to add coreutils as a dependency?

jsamr, 2019-01-31:
> Well the hash files contain the filenames along with the hashes. So we can check if a md5sums* file is relevant to the iso, and only fail if the filename is in there.
> 
> So as I've currently implemented it:
> ### Flow I : $filename.$hash exists
> 
>     1. if --no-hash-check is not set, automatically check the sum
> 
>     2. if the test fails:
>        a. if `--force-hash-check`, failAndExit
>        b. else print an echowarn message and prompt the user
> 
> 
> ### Flow II : hash candidate exists (md5sums*)
> 
>     1. if `--no-hash-check` is not set and md5sums* contains filename
> 
>     2. Do I.2
> 
> 
> ### Flow III, --hash-file set
> 
>     1. If hash file does not exists, failAndExit
> 
>     2. Do I.2 for all hash types
> 
> 
> ### Flow IV, `--force-hash-check` set
> 
>     1. After all above flows, no applicable hash is found, failAndExit
> 
> 
> If you'd like some changes to this, let me know!

Yes you are absolutely right (guess I've been lazy enough to never open a md5sum file) ; I'm good with this flow.

> Also about the dependencies:
> {"md5sum", "sha1sum", "sha256sum", "sha512sum"} reside in coreutils on debian and archlinux.
> Do we need to add coreutils as a dependency?

yes, please append commandDependencies *and* commandPackages lists, mapping all the aforementioned utilities to `coreutils`

closed 2019-04-13 by jsamr