jsamr/bootiso #19: Bootiso could allow creation of GPT partitions

jsamr, 2018-07-14:
Keeping track suggestion from @Sporif in #15.
It is not scheduled yet anytime soon; please upvote if you'd like this feature to be implemented.

dilbert-tctc, 2018-07-15:
yes please, it is a like to have feature but not a must have feature right now. thanks.

mevdschee, 2019-02-20:
Latest Windows 10 (Win10_1809Oct_EnglishInternational_x64.iso) in EUFI mode requires GPT partition table and contains a >4 GB file so it also requires an NTFS partition (and Rufus uses [UEFI:NTFS](https://github.com/pbatard/uefi-ntfs) to create a fat16 partition with EFI folder to jumpstart from)

jsamr, 2019-02-20:
@mevdschee Have you tried with [the wimsplit feature shipped in 3.2 version ?](https://github.com/jsamr/bootiso/releases/tag/v3.2.0)

mevdschee, 2019-02-20:
Wow.. I just learned about wimsplit.. I'm going to try it out and let you know!

mevdschee, 2019-02-20:
I succeeded to run the latest version, but it failed when splitting. Could it be that your iso is loop-mounted in **read-only** mode and that wim-split fails due to that? When I split manually everything works fine!

jsamr, 2019-02-20:
@mevdschee perhaps you don't have wimsplit 1.13 and you fall under #43 bug!

mevdschee, 2019-02-20:
@jsamr Thank you, you are right. That is the problem. I'm having wimtools 1.12.

mevdschee, 2019-02-20:
Just sharing my achievement/findings.

Instead of splitting the "install.wim" into parts (this is not supported in the Win10 installer) I recompressed the "install.wim" and went from 4.4Gb to 3.3Gb. The command I used was:

    $ time wimlib-imagex optimize install.wim --solid
    "install.wim" original size: 4253057 KiB
    Using LZMS compression with 4 threads
    Archiving file data: 9344 MiB of 9344 MiB (100%) done
    "install.wim" optimized size: 3175915 KiB
    Space saved: 1077141 KiB

    real	36m26,364s
    user	127m4,804s
    sys	0m10,063s

After this I was able to do an UEFI Win10 Oct 2018 (Win10_1809) install from FAT32.

jsamr, 2020-05-08:
GPT support is getting cooked for 4.0 release, get ready!

commit 00914734 (2020-05-13): implement automatic support for GPT in UEFI + Mount+Rsync

fix #19
closed 2020-05-17 by jsamr