jsamr/bootiso #21: bug report install-mount-rsync fails to find local ubuntu syslinux 6.03

dilbert-tctc, 2018-07-15:
---------- bug report bootiso 3.1.1 and thinkpad-t470 loaded with ubuntu 18.04 

install-mount-rsync fails to find local ubuntu syslinux 6.03 

---------- check versions

dilbert@thinkpad-t470:~$ bootiso -v
3.1.1
dilbert@thinkpad-t470:~$ uname -a
Linux thinkpad-t470 4.15.0-25-generic #27-Ubuntu SMP Mon Jul 2 08:35:25 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
dilbert@thinkpad-t470:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04 LTS
Release:	18.04
Codename:	bionic

---------- install-auto works 

dilbert@thinkpad-t470:~$ bootiso ubuntu-18.04-desktop-amd64.iso 
bootiso: Found hybrid ISO; choosing image copy `dd' mode.
bootiso: Partition label automatically set to `UBUNTU 18.0'.
         You can explicitly set label with `-L, --label' option.
bootiso: Listing drives available in your system:
         NAME   MODEL            VENDOR     SIZE TRAN   HOTPLUG
         sdc    Extreme          SanDisk   14.6G usb          1
bootiso: Autoselecting `sdc' (only USB device candidate)
bootiso: The selected device `/dev/sdc' is connected through USB.
bootiso: About to wipe out the content of device `/dev/sdc'.
         Are you sure you want to proceed? (y/n)>y
bootiso: Erasing contents of `/dev/sdc'...
bootiso: Copying files from ISO to USB device with `dd'    
bootiso: Synchronizing writes on device `/dev/sdc'    
bootiso: Took 56 seconds to perform [install-auto] action.
bootiso: USB device succesfully ejected.
         You can safely remove it!

---------- install-mount-rsync fails to find local syslinux

dilbert@thinkpad-t470:~$ bootiso --mrsync ubuntu-18.04-desktop-amd64.iso 
bootiso: UEFI boot check validated. Your USB will work with UEFI boot.
bootiso: Found SYSLINUX config file and binary at version 6.03.
bootiso: A SYSLINUX booloader will be installed on your USB.
bootiso: Partition label automatically set to `UBUNTU 18.0'.
         You can explicitly set label with `-L, --label' option.
bootiso: Listing drives available in your system:
         NAME   MODEL            VENDOR     SIZE TRAN   HOTPLUG
         sdc    Extreme          SanDisk   14.6G usb          1
bootiso: Autoselecting `sdc' (only USB device candidate)
bootiso: The selected device `/dev/sdc' is connected through USB.
bootiso: Created ISO mount point at `/mnt/iso.V1p'.
bootiso: About to wipe out the content of device `/dev/sdc'.
         Are you sure you want to proceed? (y/n)>y
bootiso: Erasing contents of `/dev/sdc'...
bootiso: Creating MBR partition table with `sfdisk' v2.31...
         Checking that no-one is using this disk right now ... OK
         
         Disk /dev/sdc: 14.6 GiB, 15693664256 bytes, 30651688 sectors
         Units: sectors of 1 * 512 = 512 bytes
         Sector size (logical/physical): 512 bytes / 512 bytes
         I/O size (minimum/optimal): 512 bytes / 512 bytes
         
         >>> Created a new DOS disklabel with disk identifier 0xdecf74ec.
         /dev/sdc1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 14.6 GiB.
         /dev/sdc2: Done.
         
         New situation:
         Disklabel type: dos
         Disk identifier: 0xdecf74ec
         
[... 233 lines cut ...]
         There are 32 reserved sectors.
         Volume ID is db802560, volume label F28-WORK-X8.
bootiso: Created USB device mount point at `/mnt/usb.QTJ'
bootiso: Copying files from ISO to USB device with `rsync'    
bootiso: Synchronizing writes on device `/dev/sdc'    
bootiso: Found ISOLINUX config file at `/mnt/usb.QTJ/isolinux/isolinux.cfg'.
         Moving to `/mnt/usb.QTJ/isolinux/syslinux.cfg'.
bootiso: Found local SYSLINUX version `6.03'
bootiso: ISO SYSLINUX version doesn't match local version.
         Scheduling download of version 6.04...
bootiso: Found `/var/cache/bootiso/syslinux-6.04-pre1.tar.gz' in cache.
bootiso: SYSLINUX version `6.04' temporarily set for installation.
bootiso: Installing SYSLINUX bootloader in `/mnt/usb.QTJ/isolinux' with kernel.org version `6.04'...
         /mnt/usb.QTJ/isolinux is device /dev/sdc1
bootiso: Successfully installed SYSLINUX bootloader at version `6.04'.
bootiso: Succesfully installed Master Boot Record program.
bootiso: Took 142 seconds to perform [install-mount-rsync] action.
bootiso: ISO succesfully unmounted (/mnt/iso.iKV).
bootiso: USB device partition succesfully unmounted.
bootiso: USB device succesfully ejected.
         You can safely remove it!

---------- the end 

we have needed a great cli tool like this for a long time! thanks for sharing!

jsamr, 2018-07-15:
OK, thanks for the report and the kudos!
 `extlinux: command not found` is surprising, since `extlinux` should be shipped with `syslinux`.  
I will add a check here anyway.
Can you confirm you're running Ubuntu 18.04 ?
If so, could you provide the output of `apt-file list syslinux` to see what files package `syslinux` owns?

dilbert-tctc, 2018-07-15:
dilbert@thinkpad-t470:~$ apt-file list syslinux
syslinux: /usr/bin/syslinux
syslinux: /usr/lib/SYSLINUX/altmbr.bin
syslinux: /usr/lib/SYSLINUX/gptmbr.bin
syslinux: /usr/lib/SYSLINUX/mbr.bin
syslinux: /usr/share/doc/syslinux/README.txt.gz
syslinux: /usr/share/doc/syslinux/changelog.Debian.gz
syslinux: /usr/share/doc/syslinux/copyright
syslinux: /usr/share/lintian/overrides/syslinux
syslinux: /usr/share/man/man1/syslinux.1.gz

dilbert-tctc, 2018-07-15:
dilbert@thinkpad-t470:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04 LTS
Release:	18.04
Codename:	bionic

dilbert-tctc, 2018-07-15:
dilbert@thinkpad-t470:~$ uname -a
Linux thinkpad-t470 4.15.0-25-generic #27-Ubuntu SMP Mon Jul 2 08:35:25 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

dilbert-tctc, 2018-07-15:
possible error is
rsync: change_dir "/usr/lib/syslinux/bios" failed: No such file or directory 
as my local tree is
usr/lib/syslinux/modules/bios/...

jsamr, 2018-07-15:
Thanks. I'll dig this and handle more carefully distro variations around syslinux deployment :-)

HassanMirza01, 2018-07-24:
Infected from same error, Exactly same, i modified code myself but still there is no bios folder where its looking for, install-auto, install-dd both works fine.... i think we have to use the windows's isos method on linux iso's too, i will try to add gpt support in it too, 

cool cli tool, thanks though ;)

jsamr, 2018-07-24:
@HassanMirza01 Sorry about that ; I'll check that tonight!

HassanMirza01, 2018-07-24:
not an issue, i figured it out myself, actually mbr.bin file is in lib/syslinux/mbr/ not in bios/mbr/
and also, due to this path issue, dd command also failed whiling installing mbr in usb, new bios folder is in syslinux/modules/ and not in syslinux/ ;)

HassanMirza01, 2018-07-25:
@jsamr even though i didn't understood that extbin thing but my iso of debian booting fine with correct syslinux install with install-mount-rsync option, still in burning iso, i can see errors of extbin but c32 bios modules installed successfully... i think we dont need extbin at all, i didn't find anything regarding it on web even....

jsamr, 2018-07-25:
@HassanMirza01 `extBin` is a local variable containing the file location of `extlinux` binary, which should be shipped with any `syslinux` package but apparently it is not the case on ubuntu systems. [See official wiki :-)](https://www.syslinux.org/wiki/index.php?title=EXTLINUX). Getting my hands dirty and fixing this issue this morning.

HassanMirza01, 2018-07-25:
Well, not an issue, for me, linux iso's working fine with both methods, even though extbin errors came but i will remove their lines etc, 

Am focussing on windows iso stuff now, just noticed that ms-sys discontinued long ago, find a way to get it done ;)

jsamr, 2018-07-25:
@HassanMirza01 That is probably because the ISO you're referring to supports UEFI as your BIOS do.  Syslinux job is to install a legacy BIOS bootloader (and the MBR program) to boot on older devices. You can confirm this with `bootiso --inspect my-iso-file.iso` :-) Anyway, I'm glad it works for you in that peculiar case but I must fix it anyway ^^

HassanMirza01, 2018-07-25:
my laptop support both legacy and uefi boots, but windows iso's need to support correct one according to partition scheme of hdd, in that case, its GPT, so i need uefi one, mbr is old fashioned now, i have found a way to achieve gpt stuff too... i will may be made a pull request for bootiso ;)

jsamr, 2018-07-25:
@HassanMirza01 OK, see if your experience with Windows is related to #15. For any pull requests, I suggest we discuss first some details in [the gitter channel](https://gitter.im/bootiso) I've just open on purpose!

HassanMirza01, 2018-07-25:
i succeeded in booting GPT burned on my Ubuntu 18.04, and its showing drives and i know that issue, drivers missing it says if we boot mbr on gpt or gpt on mbr etc, but i succeeded in achieving it... tried windows 10 1607 LTSB and it worked perfectly, adding in my program, actually i wrote my own script named BurnISO but i was missing some commands which i gets from your script, commands like syslinux related, i have added gpt thing in that and its working like a charm ;)

jsamr, 2018-07-25:
@dilbert-tctc @HassanMirza01 
I've committed a fix.
Could you try with the next release candidate to see if the missing files errors have disappeared ?

``` bash
curl -L https://rawgit.com/jsamr/bootiso/v3.1.2-rc-1/bootiso -O
chmod +x bootiso
```

HassanMirza01, 2018-07-25:
man, you are doing it more complicated... installing mbr is just one command thing, thats why i removed all syslinux stuff i added from your script, its not a big thing though....

jsamr, 2018-07-25:
@HassanMirza01 `SYSLINUX` is necessary for legacy boot, and is shipped in many distro ISOs.
I'm not sure what are you referring to when you mention "complicated", but there might be several aspects that can participate in that complexity:
- when local version of SYSLINUX doesn't match the ISO version, bootiso downloads the right version locally from kernel.org. This is because I have witnessed unbooting USB sticks in such situations. Such examples include openSUSE-Tumbleweed, shipped with v4.04 and elementaryOS, shipped with v4.05 (current stable is 6.03).
- SYSLINUX packaging differs from one distro to another. The "complexity" I've added in this commit with `checkSyslinuxInstall` is to find the appropriate files.

Perhaps you have now more elements in mind to understand the need for such "complexity", but if you don't, and if you find there is a better, simpler way to achieve the same results, with the same scope of compatible systems, I'm very open to hear it.

HassanMirza01, 2018-07-25:
Cool bro... I understood it well now, actually your install mbr program is too long may be thats why i responded like that, also, i never used other distros than *ubuntu,

jsamr, 2018-07-25:
@HassanMirza01 np :-)

jsamr, 2018-07-25:
@HassanMirza01 It would be very nice of you to check there are no more errors with the v3.1.2-rc-1!

closed 2018-07-26 by jsamr

jsamr, 2018-07-26:
I've tested in an Ubuntu VM and the issue did not occur. Closed.

dilbert-tctc, 2018-07-28:
I just did a quick check with Ubuntu and it worked for me. Thanks again!

commit f2888170 (2018-07-26): v3.1.2, fix #21