jsamr/bootiso #23: Windows 7 32 Bit ISO, doesn't contain .efi file

HassanMirza01, 2018-07-27:
i tried to use windows 7 sp1 32bit iso and when i inspect it, it gives,

The selected ISO is not hybrid, doesn't supports UEFI nor legacy BIOS booting with SYSLINUX.
         It cannot result in any successful booting with bootiso.

but i used rsync method forcefully and it burn iso on usb, and IT BOOTED and WORKED Fine with mbr scheme, 

can you add one more check along with ueficompatible, ishybrid and syslinuxcompatible ?? something which is present only in old windows iso's specifically windows 7 ones, 32bit, or windows vista 32bit...

jsamr, 2018-07-27:
@HassanMirza01 Thanks for your report. This is weird !
I will try to reproduce this weekend as I don't understand how is that possible ahah.

HassanMirza01, 2018-07-27:
even i tried to see iso files content, there is no .efi file, and i think windows iso's are not hybrid, syslinux also not included in windows one, so may be issue is real.... will wait for your answer...

jsamr, 2018-08-06:
@HassanMirza01 Yes, after checking this up, Windows 7 32bits doesn't support UEFI.
So I'll close this ticket and open a new one to support Windows XP and 7 in legacy boot. This is very easy with SYSLINUX if I create a config file (already done that, works great).

closed 2018-08-06 by jsamr

HassanMirza01, 2018-08-06:
but its working fine with rsync method... even with your tool... you just have to let it pass through checks so it can use rsync method... DD wont work on any windows ISO, there needed just another check which checks something in legacy ISO's of XP and 7... all fine then...

jsamr, 2018-08-06:
@HassanMirza01 It could be your BIOS which is smart enough to notice the `bootmgr` file and start the boot process from here ; but I don't think I can guarantee it will work on any motherboard. I must install the bootloader program which in turns loads SYSLINUX which then starts `bootmgr` ^^

HassanMirza01, 2018-08-06:
haha, my Main machine can't even install 32bit windows 7, i installed in on my 2nd PC, which is dell optiplex 755, a core 2 duo machine... it accept everything fine... but why it needed syslinux ?? in windows ?? you should once try with --mrsync option nd just see if your bios picks it up

jsamr, 2018-08-06:
@HassanMirza01 I will definitely do that ; SYSLINUX does a full MBR boot process, here is an extract from *How Linux Works, What every superuser should know* (great book)

> In addition to the partition information described in 4.1 Partitioning Disk Devices, the Master Boot Record (MBR) includes a small area (441 bytes) that the PC BIOS loads and executes after its Power-On Self-Test (POST). Unfortunately, this is too little storage to house almost any boot loader, so additional space is necessary, resulting in what is sometimes called a multi-stage boot loader. In this case the initial piece of code in the MBR does nothing other than load the rest of the boot loader code. The remaining pieces of the boot loader are usually stuffed into the space between the MBR and the first partition on the disk.
> Of course, this isn’t terribly secure because anything can overwrite the code there, but most boot loaders do it, including most GRUB installations. In addition, this scheme won’t work with a GPT-partitioned disk using the BIOS to boot because the GPT table information resides in the area after the MBR. (GPT leaves the traditional MBR alone for backward compatibility.)

So the MBR boot program is responsible for launching `bootmgr` binary!

HassanMirza01, 2018-08-06:
but not syslinux... syslinux is for Linux stuff and MBR & GPT for windows isos... i didn't knew that in detail but your right now...

jsamr, 2018-08-06:
> chain.c32 is a COM32 module for Syslinux. It can chainload MBRs, partition boot sectors, **Windows bootloaders (ntldr, setupldr.bin and bootmgr)**, **MS-DOS** and **PC-DOS** io.sys, Freedos kernel.sys, isolinux.bin (only from ISOLINUX), grldr of grub4dos, or a bootsector saved to a file. It can also swap BIOS drive numbers, or hide partitions (via partition's ID). 

@HassanMirza01 Yes, SYSLINUX has support for legacy Windows, [check the chain.c32 module](https://www.syslinux.org/wiki/index.php?title=Comboot/chain.c32) which has a `ntldr` option ;-)

HassanMirza01, 2018-08-06:
thats gonna be good then... well, my tool supports mine pcs, thats enough for me atleast for now... now i will create data partition after dd command, and i will use dd commands for linux and rsync for windows...