jsamr/bootiso #24: It gonna destroy a External HDD connected with USB

HassanMirza01, 2018-08-03:
i saw once in rufus log that it detects somehow with some digits (i will check it again how) that if a usb is hdd or not, if its hdd, it wont use or name that drive, but i just saw that bootiso showing me the external hdd as usb device and it wont resist in formatting it, its 2TB Data, i suggest to add a check in selectDrive to see difference in udev properties of a external hdd and normal usb...

HassanMirza01, 2018-08-03:
Listing USB devices available in your system:
         
         NAME   MODEL            VENDOR     SIZE TRAN   HOTPLUG
         sdb    Expansion        Seagate    1.8T usb          1

jsamr, 2018-08-03:
@HassanMirza01 OK ! Would you mind checking where we can get this information ?

HassanMirza01, 2018-08-03:
sure... i have to get a log for that... download kali linux and will burn it with rufus, will tell whats different in external hdd and normal usb,,,, actually i can't risk 2TB data loss in anyway, so am resisting to check it via any script... will check with udev comamands etc... will reply when i got that ;)

HassanMirza01, 2018-08-03:
FIXED!!
https://github.com/HassanMirza01/bootiso/commit/11803a7d45e17594c12c4e300c038a25534cadb4

jsamr, 2018-08-03:
@HassanMirza01 That's very cool ! Thanks. I'll add an "exclude non-thumb drives" policy by default in next release thanks to your findings. Along with an `--include-non-thumbs`, `-N` flag to override this behaviour.

HassanMirza01, 2018-08-05:
bro, got one more issue... i prefer rsync method instead of dd, may be thats why i get most of the issues... i use dd method on latest kali linux and it worked fine, it created 2 partitions, one for mbr and syslinux files nd other with kali linux data files... but when i did same with rsync, it made it fine, and even my PC gets it booted but while installing, i saw just debian like stuff only, that was not kali linux, it says mount cd failed, it stuck there... but with dd, same iso booted and installed fine... whats that ?? splash was of kali linux... am sending you the output of program and files screenshot in a moment...

HassanMirza01, 2018-08-05:
`hassanmirza01@bionic:~$ ./bootiso --mrsync '/home/hassanmirza01/Downloads/kali-linux-rolling-amd64.iso'
bootiso: UEFI boot check validated. Your USB will work with UEFI boot.
bootiso: Found SYSLINUX config file and binary at version 6.03.
bootiso: A SYSLINUX booloader will be installed on your USB.
bootiso: Partition label automatically set to `KALI LIVE'.
         You can explicitly set label with `-L, --label' option.
bootiso: Listing drives available in your system:
         NAME   MODEL            VENDOR     SIZE TRAN   HOTPLUG SERIAL
         sdb    Transcend 8GB    JetFlash   7.6G usb          1 026HATT95ZEEN9E2
bootiso: Autoselecting `sdb' (only USB device candidate)
bootiso: The selected device `/dev/sdb' is connected through USB.
bootiso: Created ISO mount point at `/mnt/iso.9Bv'.
bootiso: About to wipe out the content of device `/dev/sdb'.
         Are you sure you want to proceed? (y/n)>y
bootiso: Erasing contents of `/dev/sdb'...
bootiso: Creating MBR partition table with `sfdisk' v2.31...
         Checking that no-one is using this disk right now ... OK
         
         Disk /dev/sdb: 7.6 GiB, 8103395328 bytes, 15826944 sectors
         Units: sectors of 1 * 512 = 512 bytes
         Sector size (logical/physical): 512 bytes / 512 bytes
         I/O size (minimum/optimal): 512 bytes / 512 bytes
         
         >>> Created a new DOS disklabel with disk identifier 0x5f0fcab8.
         /dev/sdb1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of size 7.6 GiB.
         /dev/sdb2: Done.
         
         New situation:
         Disklabel type: dos
         Disk identifier: 0x5f0fcab8
         
         Device     Boot Start      End  Sectors  Size Id Type
         /dev/sdb1  *     2048 15826943 15824896  7.6G  c W95 FAT32 (LBA)
         
         The partition table has been altered.
         Calling ioctl() to re-read partition table.
         Syncing disks.
bootiso: Creating vfat partition on `/dev/sdb1'...
         mkfs.fat 4.1 (2017-01-24)
         /dev/sdb1 has 250 heads and 62 sectors per track,
         hidden sectors 0x0800;
         logical sector size is 512,
         using 0xf8 media descriptor, with 15824896 sectors;
         drive number 0x80;
         filesystem has 2 32-bit FATs and 8 sectors per cluster.
         FAT size is 15424 sectors, and provides 1974252 clusters.
         There are 32 reserved sectors.
         Volume ID is 98ff0541, volume label KALI LIVE  .
bootiso: Created USB device mount point at `/mnt/usb.6Lo'
bootiso: Copying files from ISO to USB device with `rsync'    
bootiso: Synchronizing writes on device `/dev/sdb'    
bootiso: Found ISOLINUX config file at `/mnt/usb.6Lo/isolinux/isolinux.cfg'.
         Moving to `/mnt/usb.6Lo/isolinux/syslinux.cfg'.
bootiso: Found local SYSLINUX version `6.03'
bootiso: ISO SYSLINUX version matches local version.
bootiso: Installing SYSLINUX bootloader in `/mnt/usb.6Lo/isolinux' with local version `6.03'...
bootiso: C32 BIOS modules successfully installed.
         /mnt/usb.6Lo/isolinux is device /dev/sdb1
bootiso: Successfully installed SYSLINUX bootloader at version `6.03'.
bootiso: Succesfully installed Master Boot Record program.
bootiso: Took 628 seconds to perform [install-mount-rsync] action.
bootiso: ISO succesfully unmounted (/mnt/iso.9Bv).
bootiso: USB device partition succesfully unmounted.
bootiso: USB device succesfully ejected.
         You can safely remove it!
hassanmirza01@bionic:~$ `
Thats full program...

jsamr, 2018-08-05:
@HassanMirza01 `mount+rsync` is less reliable because sometimes, distros include multiple partitions which this mode won't exactly reproduce.
You can confirm this with `fdisk -l '/home/hassanmirza01/Downloads/kali-linux-rolling-amd64.iso'
`.
I downloaded kali and found that there is a `FAT12` partition shipped. I don't know the exact purpose but this is definitly the reason it's not working properly in mount+rsync mode.

``` text
jsamr@localhost ~  $ fdisk -l ~/Downloads/Torrents/kali-linux-2018.2-amd64/kali-linux-2018.2-amd64.iso 
Disque /home/jsamr/Downloads/Torrents/kali-linux-2018.2-amd64/kali-linux-2018.2-amd64.iso : 2.8 GiB, 2936733696 octets, 5735808 secteurs
Unités : secteur de 1 × 512 = 512 octets
Taille de secteur (logique / physique) : 512 octets / 512 octets
taille d'E/S (minimale / optimale) : 512 octets / 512 octets
Type d'étiquette de disque : dos
Identifiant de disque : 0xa9ef3e16

Périphérique                                                                        Amorçage   Début     Fin Secteurs Taille Id Type
/home/jsamr/Downloads/Torrents/kali-linux-2018.2-amd64/kali-linux-2018.2-amd64.iso1 *             64 5734399  5734336   2.8G 17 HPFS/NTFS masquée
/home/jsamr/Downloads/Torrents/kali-linux-2018.2-amd64/kali-linux-2018.2-amd64.iso2          5734400 5735807     1408   704K  1 FAT12
```

dpserver, 2018-08-06:
Good news, not issue :)
Utilities: Gufw, Konsole, Ark, Image Writer, Bootiso...
https://septor.sourceforge.io

jsamr, 2018-08-06:
@HassanMirza01 The trick is that to have access to the second+ partitions, you must mount them with `offset` option.
eg:

```
mount -o loop,offset=$((5734400*512)) '/home/hassanmirza01/Downloads/kali-linux-rolling-amd64.iso'
```
To find the good offset value, you multiply the first partition sector (5734400) with sector size.
I have no idea why it would work with GPT, since partition table scheme seems orthogonal to this issue.

HassanMirza01, 2018-08-06:
Actually i tried both dd and rsync methods with GPT while installing my main OS as my machine is GPT schemed... I thought mbr gonna be same and i just settled it.. then i had to install an OS on my 2nd PC, which is MBR, i used rsync method but then i report to you, actually with DD on GPT, it still write one partition and even i didn't needed to set the partition with boot flag, it works fine with GPT as efi files are present in cd or 2nd partition of big size, first small partition contains only mbr related files AFAIK...

jsamr, 2018-08-06:
@dpserver Thanks a lot for the info! I'm glad it gets shipped in a distro ;-)
@HassanMirza01 `dd` will erase the partition table scheme along with any partition, since it copies the entire ISO which contain those... That is why `dd` will work where mount+rsync don't. If I understand you correctly, the installation thumb media works with dd+gpt (but as I told you, GPT should be irrelevant) while it doesn't with rsync+mbr. Given the output of `fdisk` it looks like the first partition is the fat one (2.8GB) while the second is the small ESP (EFI System Partition)... Well I might have misunderstood something, please apologies if that's the case ^^

HassanMirza01, 2018-08-06:
Haha, if something working fine, i dont even want to understand iy :D but i know the basics like its been 12years am using Computers... Started when windows 98 was there... 

DD is good choice and now i got another idea of getring it... Why i want rsync is due to availability of additional disk space, i will make like a partition sdb1 of about 10gb out of 16gb, and will install dd on 4gb space (14-10)gb... Hows the idea ??

jsamr, 2018-08-06:
@HassanMirza01 You should be able to create a new partition with the whole remaining space after DD has occurred. `fdisk` will get you there :-) The script will do this in next release, see #1 and `-T` flag.

HassanMirza01, 2018-08-06:
Actually am running my own script which follow stuff like yours, but am using sgdiak for zapping and like that,  not a fan of arguments things so am using what i prefer, if thats possible, i will do it with parted, i prefer parted as clinof gparted

jsamr, 2018-08-06:
@HassanMirza01 `parted` supports both GPT and MBR? When bootiso come to support GPT, it could be a nice move switching to `parted` because `sfdisk` and `sgdisk` have very different interfaces. So if you do, I'll take a look at your fork and get some inspiration there ;-) I'll add you to a list of contributors also.

HassanMirza01, 2018-08-06:
not an issue for me... actually i decided to release my tool but when i saw yours, i thought there is already a good one there so i should support that as i never found a valid working script when it comes to burn iso, i always turn on windows(dual boot) to use rufus.. i then decided to make my own, 

and Yes, parted supports both gpt and mbr, it can make partitions, tables, labels, everything, am using all its commands, just do parted -s $devicename mktable msdos/gpt and it will create table scheme on a wiped disk... its all in one tool man, hope so you have used GParted tool, it can do all stuff which gparted can do.... i extract commands from gparted too, like formatting in ext4,ext3, ext2,ntfs,exfat,f2fs, and integrated stuff in my tool... now am gonna add Zenity in it so i dont have to drag and drop iso on terminal when it ask to provide iso...

jsamr, 2019-01-30:
@HassanMirza01 Thanks for your suggestion and commit ; @SibrenVasse has integrated this in `next` branch.

closed 2019-01-30 by jsamr

jsamr, 2019-04-04:
For your information, this feature has been abandoned. The reason is that the information returned by  `udevadm` is extracted from a hardcoded whitelist and therefore not consistent. You can see al the discussion in PR #40

PR #39 (2019-01-30): USB thumb drive check
Fixes #24 

Adapted from 11803a7 to incorporate all command line options etc. Changed the command line option from your suggestion in the issue to --no-thumb-check, which I found to be more fitting. Do you agree?

Line 17 is a change made by shfmt.