jsamr/bootiso #25: Bootiso should support Windows 7 32bits and Windows XP

jsamr, 2018-08-06:
Those OS only work in legacy boot ; which can be installed with SYSLINUX.
To do so, one config file must be created which tells SYSLINUX to chainload the `bootmgr` file.

HassanMirza01, 2018-10-08:
i just added support for it.... just let it go with mbr... remove failandexit variable and use a read command with default "y" option... that way, it will install syslinux mbr compatible and that will work with windows 7 isos... like it should ask if you still wanna proceed at your own risk ?? y/n: y
and if it says yes, just override the failandexit and put it in "n" input ;)

HassanMirza01, 2018-10-08:
[image]

closed 2020-10-17 by jsamr