jsamr/bootiso #26: MINIX 3 ISO Support

alxersov, 2018-08-17:
Is it possible to add support for MINIX3 ISO images?

I get the following error for the [current version](http://download.minix3.org/iso/snapshot/minix_R3.4.0rc6-d5e4fc0.iso.bz2):

```
bootiso -i minix_R3.4.0rc6-d5e4fc0.iso

bootiso: The selected ISO is not hybrid, doesn't supports UEFI nor legacy BIOS booting with SYSLINUX.
         It cannot result in any successful booting with bootiso.
```

Thanks,
Alexei

jsamr, 2018-08-17:
@alxersov Unfortunately, the message claim is true. As per [the official website](https://wiki.minix3.org/doku.php?id=usersguide:doinginstallation):

>  This [guide](https://wiki.minix3.org/doku.php?id=usersguide:x86usbworkaround) is a workaround to install MINIX on Intel PC using USB. **There is no actual USB support**, use it at your own risk. 

And the provided guide involves using a virtual machine! Way too complex... Plus I would have to add a dependency on QEMU. I'm sorry, you'll have to get your hands dirty :-(

closed 2018-08-17 by jsamr

alxersov, 2018-08-18:
@jsamr Thank you! So SYSLINUX does not support this ISO, but why? 🤔 What is wrong with the image? It is possible to [boot MINIX over network using PXE](https://wiki.minix3.org/doku.php?id=usersguide:bootingovernetworkusingpxe). If PXELINUX works doesn't this mean that SYSLINUX should also support MINIX? I am a little confused and I'd like to better understand how it all works and where the limitation comes from.

jsamr, 2018-08-18:
@alxersov  ~~The main issue with this image is that [it's not hybrid](https://superuser.com/questions/683210/how-do-i-determine-if-an-iso-is-a-hybrid). **Hybrid ISOs include a full disk image encompassing a partition table (usually MBR) and a boot mechanism, such as UEFI compliant partition or legacy BIOS bootloader**~~.

While I was typing, I made a test.
And actually, the latest *was* hybrid!
[image]

Can you try again? I doubt though it will result in a successful boot process since I mounted the ISO and made a few checks.

### I. Found thoses lines in the readme:

```
HARDWARE
*MINIX 3 OS requires the following hardware:
        - PC with a Pentium or compatible processor
        - at least 200 MB of free disk space
        - IDE CD-ROM drive
        - IDE hard disk
        - NOT SUPPORTED: USB and SCSI disks; Some Serial ATA disks work.
```

### II. There is no UEFI partition ([ESP](https://wiki.archlinux.org/index.php/EFI_system_partition))

```
$ fdisk -l minix_R3.3.0-588a35b.iso 
Disque minix_R3.3.0-588a35b.iso : 577.5 MiB, 605581312 octets, 1182776 secteurs
Unités : secteur de 1 × 512 = 512 octets
Taille de secteur (logique / physique) : 512 octets / 512 octets
taille d'E/S (minimale / optimale) : 512 octets / 512 octets
Type d'étiquette de disque : dos
Identifiant de disque : 0x00000000

Périphérique              Amorçage Début     Fin Secteurs Taille Id Type
minix_R3.3.0-588a35b.iso1              0   12287    12288     6M 81 Minix / Linux ancien
minix_R3.3.0-588a35b.iso2          12288   82087    69800  34.1M 81 Minix / Linux ancien
minix_R3.3.0-588a35b.iso3          82088 1182775  1100688 537.5M 81 Minix / Linux ancien
```

alxersov, 2018-08-19:
Thanks for checking. I've tried `minix_R3.3.0-588a35b.iso` image. `bootiso` processed this image with no error, but the PC would not boot from USB ("No valid operating system" error).

jsamr, 2018-08-19:
@alxersov The good news is that the ISO I checked is an hybrid ISO, while the one you reported wasn't. I bet MINIX maintainers are on the way to make it bootable. For such thing to happen, they would need to add MBR bootloader program. If you want more information on this boot process, [check wikipedia entry for MBR](https://en.wikipedia.org/wiki/Master_boot_record) :-).