jsamr/bootiso #27: bootiso is 'listing drives' as 'lsblk: unknown column' (??)

RonCam, 2018-08-18:
I have partitioned a 16MiB flash drive into, approximately, two 8MiB partitions, 'boot' (with boot flag set) and 'storage'.  Both are FAT-32, primary partitions, as follows:
```
/dev/sdc1: LABEL="BOOT" UUID="12A2-C2CD" TYPE="vfat"
/dev/sdc2: LABEL="STORAGE" UUID="1379-2FEB" TYPE="vfat"
```
However, when _bootiso_ is run, there is a problem in the last two lines of the terminal output:
```
 > ./bootiso -p tails-amd64-3.8.iso
bootiso: Reporting `tails-amd64-3.8.iso' boot capabilities:
bootiso: The selected ISO is hybrid and supports legacy BIOS booting with SYSLINUX 6.03 along with UEFI boot.
         It will result in successful booting with bootiso in [install-auto] and [install-dd] modes with any PC BIOS.                                                                                                                  
bootiso: Listing drives available in your system:
lsblk: unknown column: VENDOR,SIZE,TRAN,HOTPLUG,SERIAL
lsblk: unknown column: VENDOR,SIZE,TRAN,HOTPLUG,SERIAL
```
Is it possible to do something with command line parameters to direct _bootiso_ to use `/dev/sdc1` as the target -- in spite of it not being automatically detected?  Any other suggestions would be most appreciated!

jsamr, 2018-08-18:
@RonCam `bootiso` is not aimed at doing such things.
The reason is that hybrid ISOs are disk images, containing not only one to many partitions, but also partition tables and so on.
However I see you are trying to add a storage partition.
You can easily do this already! I checked the ISO, it will be defaulted to `dd-install` mode in bootiso.
After you have "burned" the USB with `bootiso`, just use your favorite MBR partition wizard (`fdisk`, `parted`) and add a vfat storage partition on the remaining space. Make the filesystem (`mkfs`) and you're good to go :-)

RonCam, 2018-08-18:
Many thanks for a fast and to-the-point reply.  So, there is a work-around!

If I understand correctly, the procedure is to start off as if you wanted to use the _entire_ flash drive as a boot device, then add the second, third, or whatever number of _vfat_ (or _ext_?) storage partitions, that are wanted.

It's something I'll have to try and see how it goes, because -- and, apparently this may not be the case, because you're saying it can be done -- I thought, when you add a partition to a drive, you have to redo the partition table, which then wipes out anything else that's already on the drive.  I'm just going by my memory of what I recall from the Menu options offered in _gparted_, that I saw while preparing this drive.  

And, you are correct, no, I was not trying to make a multi-boot drive.  Just trying not to waste the extra multiple-GiBs left on the drive.

RonCam, 2018-08-18:
Ah, now I think I understand.  

It may be done by _resizing_ the existing partition to create the new space, not by _creating_ an additional partition.  OK, that sounds better!

HassanMirza01, 2018-08-19:
@jsamr i was busy a bit in my professional life, i find the way to your issue no.1 i.e persistent data partition, i wrote the code for it, but i wanna see its versatility so there comes no error in diff iso's and usb's...