jsamr/bootiso #28: Error message from gparted when trying to resize newly-created boot partition

RonCam, 2018-08-19:
I am following this advice:
You can easily do this already! I checked the ISO, it will be defaulted to dd-install mode in bootiso.
After you have "burned" the USB with bootiso, just use your favorite MBR partition wizard (fdisk, parted) and add a vfat storage partition on the remaining space. Make the filesystem (mkfs) and you're good to go :-)

I'm getting a bootable flash drive, but then when I go to gparted to make space for a storage partition, I'm seeing this:

> Invalid partition table - recursive partition on /dev/sdc.

I don't think the .iso is defective only because it passed the automatic verification step, with the Firefox Tails add-on.  I'm unable to proceed from this point with gparted, and so, I've never been able to create the storage partition.

Could this indicate a problem?  Note there are two drives shown (there should be only one) and the one drive isn't being detected:
```
bootiso: Listing drives available in your system:
lsblk: **unknown column**: VENDOR,SIZE,TRAN,HOTPLUG,SERIAL
lsblk: **unknown column**: VENDOR,SIZE,TRAN,HOTPLUG,SERIAL
```
The only way it works is if I use this in the command line:
`> bootiso -d sdc tails-amd64-3.8.iso`
where /dev/sdc is the target USB drive.

RonCam, 2018-08-19:
I am searching for other possible causes for what I am observing (as listed in previous post) and [I have come across this, as possibly relevant:  it could be the PNY USB drive.](https://github.com/pbatard/rufus/wiki/FAQ#list-of-hardware-and-software-rufus-isnt-compatible-with)
In the linked FAQ page, scroll down to where you see "PNY 16GB Retract USB (and some other PNY drives)" and following this, there is a link to the PNY site where "USB Flash Driver Format Tool[s]" are offered for download.
[The drive is a few years old and can't be exactly identified](https://i.ebayimg.com/thumbs/images/m/mohGy-3Nx_Os2C8iEy9SUDQ/s-l225.jpg), but unless you look at the above and say the errors look familiar, then I suspect the drive.

Any other ideas?

jsamr, 2018-08-19:
@RonCam I understand that your version of `lsblk` does not support some columns. Can you provide me info on your `lsblk --version`? I will investigate which columns are supported in your distro.

Ideally, I should try to reproduce your partition issue locally. Please give me the following:

- `bootiso --version`
- a URL to the ISO you're trying to burn (guess it's still tails)

I will try on my workstation and send you some feedback.

RonCam, 2018-08-20:
OK, thanks so much for looking into this.  
When the first command is entered in terminal, I get a very unexpected response, completely unrelated to the version number, but giving details on its column support:
```
~ > lsblk --version
lsblk: unrecognized option '--version'

Usage:
 lsblk [options] [<device> ...]

Options:
 -a, --all            print all devices
 -b, --bytes          print SIZE in bytes rather than in human readable format
 -d, --nodeps         don't print slaves or holders
 -D, --discard        print discard capabilities
 -e, --exclude <list> exclude devices by major number (default: RAM disks)
 -f, --fs             output info about filesystems
 -h, --help           usage information (this)
 -i, --ascii          use ascii characters only
 -m, --perms          output info about permissions
 -l, --list           use list format ouput
 -n, --noheadings     don't print headings
 -o, --output <list>  output columns
 -P, --pairs          use key="value" output format
 -r, --raw            use raw output format
 -t, --topology       output info about topology
Available columns:
       NAME  device name
      KNAME  internal kernel device name
    MAJ:MIN  major:minor device number
     FSTYPE  filesystem type
 MOUNTPOINT  where the device is mounted
      LABEL  filesystem LABEL
       UUID  filesystem UUID
         RO  read-only device
         RM  removable device
      MODEL  device identifier
       SIZE  size of the device
      STATE  state of the device
      OWNER  user name
      GROUP  group name
       MODE  device node permissions
  ALIGNMENT  alignment offset
     MIN-IO  minimum I/O size
     OPT-IO  optimal I/O size
    PHY-SEC  physical sector size
    LOG-SEC  logical sector size
       ROTA  rotational device
      SCHED  I/O scheduler name
    RQ-SIZE  request queue size
       TYPE  device type
   DISC-ALN  discard alignment offset
  DISC-GRAN  discard granularity
   DISC-MAX  discard max bytes
  DISC-ZERO  discard zeroes data
For more information see lsblk(8).
```
```
~ > bootiso --version
3.1.3
```
Tails download URL:
https://tails.bgadmin.com/tails/stable/tails-amd64-3.8/tails-amd64-3.8.iso

Thanks again for taking the time to help with this!  Anything else, please ask.

RonCam, 2018-08-20:
You could be correct and maybe it's not the drive after all.  Two other USB flash drives (not PNY) were selected at random and _bootiso_ was run, with the _list-usb-drives_ parameter:
First flash drive
```
~/Downloads/Tails-work-area > bootiso -l
bootiso: Listing drives available in your system:
lsblk: unknown column: VENDOR,SIZE,TRAN,HOTPLUG,SERIAL
lsblk: unknown column: VENDOR,SIZE,TRAN,HOTPLUG,SERIAL
```
Second flash drive
```
~/Downloads/Tails-work-area > bootiso -l
bootiso: Listing drives available in your system:
lsblk: unknown column: VENDOR,SIZE,TRAN,HOTPLUG,SERIAL
lsblk: unknown column: VENDOR,SIZE,TRAN,HOTPLUG,SERIAL
```
These drives are not being detected, either.  But again, a _single_ USB drive plugged into the port is generating not one, but _two_ lines, as above.

I wonder what could be the remedy for this, since the system is kept updated from the Mint repos.  I would have thougt that if _lsblk_ were buggy, it would have been automatically updated with a good copy.  But, possibly not?

RonCam, 2018-08-20:
Additional system information, possibly relevant:
```
> uname -r
3.13.0-100-generic
```
Linux Mint, as a matter of [policy](https://segfault.linuxmint.com/2013/11/answering-controversy-stability-vs-security-is-something-you-configure/), excludes routine kernel updates, but the user may choose to override the update manager settings, when the need arises. 

[This thread](https://bbs.archlinux.org/viewtopic.php?id=188178) is about four years old, but it does relate to a problem with _lsblk_ not working as expected.

jsamr, 2018-08-20:
@RonCam The error from lsblk just state that some of the columns I'm asking it to display are not supported.
I guess you're running an old version of lsblk, but I'm not sure how old (mint 17 doesn't seems too old to me).
lsblk is shipped with [util-linux package](https://packages.debian.org/stretch/util-linux).
You can check package version with a `dpkg -L util-linux`, please paste the output!

You can also monkey-patch bootiso script file and replace line 520:

``` bash
	typeset lsblkCmd='lsblk -o NAME,MODEL,VENDOR,SIZE,TRAN,HOTPLUG,SERIAL'
```
with:
``` bash
	typeset lsblkCmd='lsblk -o NAME,MODEL'
```
The error should disappear :-)

RonCam, 2018-08-20:
@jsamr Thanks again for the continued help.  Here is the output for the command you gave me:
```
~ > dpkg -L util-linux
/.
/usr
/usr/share
/usr/share/doc
/usr/share/doc/util-linux
/usr/share/doc/util-linux/README.poeigl.gz
/usr/share/doc/util-linux/README.fdisk.gz
/usr/share/doc/util-linux/README.Debian.hwclock
/usr/share/doc/util-linux/AUTHORS.gz
/usr/share/doc/util-linux/examples
/usr/share/doc/util-linux/examples/fstab.example2
/usr/share/doc/util-linux/examples/getopt-parse.bash
/usr/share/doc/util-linux/examples/sfdisk.examples.gz
/usr/share/doc/util-linux/examples/getopt-parse.tcsh
/usr/share/doc/util-linux/README.cfdisk
/usr/share/doc/util-linux/README.modems-with-agetty
/usr/share/doc/util-linux/copyright
/usr/share/info
/usr/share/man
/usr/share/man/man3
/usr/share/man/man5
/usr/share/man/man1
/usr/share/man/man1/ddate.1.gz
/usr/share/man/man1/chkdupexe.1.gz
/usr/share/man/man1/ipcmk.1.gz
/usr/share/man/man1/chrt.1.gz
/usr/share/man/man1/whereis.1.gz
/usr/share/man/man1/namei.1.gz
/usr/share/man/man1/unshare.1.gz
/usr/share/man/man1/pg.1.gz
/usr/share/man/man1/fallocate.1.gz
/usr/share/man/man1/dmesg.1.gz
/usr/share/man/man1/line.1.gz
/usr/share/man/man1/flock.1.gz
/usr/share/man/man1/getopt.1.gz
/usr/share/man/man1/linux64.1.gz
/usr/share/man/man1/lscpu.1.gz
/usr/share/man/man1/taskset.1.gz
/usr/share/man/man1/ionice.1.gz
/usr/share/man/man1/ipcrm.1.gz
/usr/share/man/man1/setsid.1.gz
/usr/share/man/man1/setterm.1.gz
/usr/share/man/man1/rev.1.gz
/usr/share/man/man1/readprofile.1.gz
/usr/share/man/man1/rename.ul.1.gz
/usr/share/man/man1/mcookie.1.gz
/usr/share/man/man1/linux32.1.gz
/usr/share/man/man1/tailf.1.gz
/usr/share/man/man1/more.1.gz
/usr/share/man/man1/ipcs.1.gz
/usr/share/man/ru/man1/ddate.1.gz
/usr/share/man/man8
/usr/share/man/man8/mkswap.8.gz
/usr/share/man/man8/partx.8.gz
/usr/share/man/man8/fstrim-all.8.gz
/usr/share/man/man8/addpart.8.gz
/usr/share/man/man8/switch_root.8.gz
/usr/share/man/man8/resizepart.8.gz
/usr/share/man/man8/isosize.8.gz
/usr/share/man/man8/mkfs.minix.8.gz
/usr/share/man/man8/rtcwake.8.gz
/usr/share/man/man8/hwclock.8.gz
/usr/share/man/man8/ldattach.8.gz
/usr/share/man/man8/raw.8.gz
/usr/share/man/man8/fdformat.8.gz
/usr/share/man/man8/blkid.8.gz
/usr/share/man/man8/agetty.8.gz
/usr/share/man/man8/mkfs.bfs.8.gz
/usr/share/man/man8/wipefs.8.gz
/usr/share/man/man8/cfdisk.8.gz
/usr/share/man/man8/getty.8.gz
/usr/share/man/man8/i386.8.gz
/usr/share/man/man8/sfdisk.8.gz
/usr/share/man/man8/fstrim.8.gz
/usr/share/man/man8/fsfreeze.8.gz
/usr/share/man/man8/findfs.8.gz
/usr/share/man/man8/cytune.8.gz
/usr/share/man/man8/fsck.8.gz
/usr/share/man/man8/fdisk.8.gz
/usr/share/man/man8/setarch.8.gz
/usr/share/man/man8/pivot_root.8.gz
/usr/share/man/man8/mkfs.8.gz
/usr/share/man/man8/lsblk.8.gz
/usr/share/man/man8/delpart.8.gz
/usr/share/man/man8/ctrlaltdel.8.gz
/usr/share/man/man8/blockdev.8.gz
/usr/share/man/man8/tunelp.8.gz
/usr/share/man/man8/fsck.minix.8.gz
/usr/share/man/man8/swaplabel.8.gz
/usr/share/lintian
/usr/share/lintian/overrides
/usr/share/lintian/overrides/util-linux
/usr/sbin
/usr/sbin/readprofile
/usr/sbin/cytune
/usr/sbin/ldattach
/usr/sbin/rtcwake
/usr/sbin/tunelp
/usr/sbin/fdformat
/usr/lib
/usr/lib/mime
/usr/lib/mime/packages
/usr/lib/mime/packages/util-linux
/usr/lib/i386-linux-gnu
/usr/bin
/usr/bin/resizepart
/usr/bin/mcookie
/usr/bin/delpart
/usr/bin/namei
/usr/bin/lscpu
/usr/bin/fallocate
/usr/bin/chkdupexe
/usr/bin/taskset
/usr/bin/ipcmk
/usr/bin/unshare
/usr/bin/partx
/usr/bin/flock
/usr/bin/setarch
/usr/bin/ipcrm
/usr/bin/getopt
/usr/bin/line
/usr/bin/chrt
/usr/bin/ionice
/usr/bin/ipcs
/usr/bin/setsid
/usr/bin/whereis
/usr/bin/addpart
/usr/bin/pg
/usr/bin/rev
/usr/bin/rename.ul
/usr/bin/ddate
/usr/bin/setterm
/etc
/etc/init.d
/etc/default
/etc/init
/etc/init/hwclock.conf
/etc/init/hwclock-save.conf
/etc/fstab.d
/etc/cron.weekly
/etc/cron.weekly/fstrim
/sbin
/sbin/cfdisk
/sbin/fsck.cramfs
/sbin/fdisk
/sbin/hwclock
/sbin/fsfreeze
/sbin/wipefs
/sbin/ctrlaltdel
/sbin/mkswap
/sbin/switch_root
/sbin/findfs
/sbin/agetty
/sbin/getty
/sbin/fstrim-all
/sbin/fsck.minix
/sbin/mkfs.minix
/sbin/blkid
/sbin/isosize
/sbin/mkfs
/sbin/mkfs.bfs
/sbin/fstrim
/sbin/sfdisk
/sbin/swaplabel
/sbin/raw
/sbin/blockdev
/sbin/pivot_root
/sbin/fsck
/sbin/mkfs.cramfs
/lib
/lib/udev
/lib/udev/rules.d
/lib/i386-linux-gnu
/bin
/bin/tailf
/bin/more
/bin/dmesg
/bin/lsblk
/usr/share/doc/util-linux/changelog.Debian.gz
/usr/share/man/man8/clock.8.gz
/usr/bin/linux64
/usr/bin/i386
/usr/bin/linux32
```
I'm pasting it in its entirety, as I can't find any mention of a 'version number'. 

After the drive has been (virtually!) 'burned' could you please open it in _gparted_, not so much for resizing, but just to see if you get the same error message, as did I -- or, does the drive open correctly in _parted_, on your system?

RonCam, 2018-08-20:
@jsamr The following is copied from Synaptic, for _util-linu,_ where it is listed _as_ 'installed':
_util-linux_2.20.1-5.1ubuntu20.9_

jsamr, 2018-08-21:
@RonCam Sorry I haven't used debian-based system for a long time... So I wasn't totally certain for dpkg ; guess it's rather the "info" command you want (perhaps `dpkg -I util-linux` ?).

Anyway, now I have its version. [Which I checked, and it happened it si from **October 2011**](https://github.com/karelzak/util-linux/releases/tag/v2.20.1).
I am very surprised it is that old ! 

### You version
So I checked the lsblk history, and found that current program has the following columns declaration:

``` c
enum {
	COL_NAME = 0,
	COL_KNAME,
	COL_PATH,
	COL_MAJMIN,
	COL_FSTYPE,
	COL_TARGET,
	COL_LABEL,
	COL_UUID,
	COL_PTUUID,
	COL_PTTYPE,
	COL_PARTTYPE,
	COL_PARTLABEL,
	COL_PARTUUID,
	COL_PARTFLAGS,
	COL_RA,
	COL_RO,
	COL_RM,
	COL_HOTPLUG,
	COL_MODEL,
	COL_SERIAL,
	COL_SIZE,
	COL_STATE,
	COL_OWNER,
	COL_GROUP,
	COL_MODE,
	COL_ALIOFF,
	COL_MINIO,
	COL_OPTIO,
	COL_PHYSEC,
	COL_LOGSEC,
	COL_ROTA,
	COL_SCHED,
	COL_RQ_SIZE,
	COL_TYPE,
	COL_DALIGN,
	COL_DGRAN,
	COL_DMAX,
	COL_DZERO,
	COL_WSAME,
	COL_WWN,
	COL_RAND,
	COL_PKNAME,
	COL_HCTL,
	COL_TRANSPORT,
	COL_SUBSYS,
	COL_REV,
	COL_VENDOR,
	COL_ZONED,
};
```
Versus (in your version)

``` c
enum {
	COL_NAME = 0,
	COL_KNAME,
	COL_MAJMIN,
	COL_FSTYPE,
	COL_TARGET,
	COL_LABEL,
	COL_UUID,
	COL_PARTLABEL,
	COL_PARTUUID,
	COL_RA,
	COL_RO,
	COL_RM,
	COL_MODEL,
	COL_SIZE,
	COL_STATE,
	COL_OWNER,
	COL_GROUP,
	COL_MODE,
	COL_ALIOFF,
	COL_MINIO,
	COL_OPTIO,
	COL_PHYSEC,
	COL_LOGSEC,
	COL_ROTA,
	COL_SCHED,
	COL_RQ_SIZE,
	COL_TYPE,
	COL_DALIGN,
	COL_DGRAN,
	COL_DMAX,
	COL_DZERO,
	COL_WWN,
};
```

The missing columns are indeed the following:

``` c
    COL_HOTPLUG,
    COL_SERIAL,
    COL_VENDOR,
    COL_TRANSPORT,
```

### Commit history

In commit history, I found those entries:

```
483987c275e461f4118a89d3be013698845d702b, 2014-12-02, lsblk: add HOTPLUG column
460c7afb79075bd5b39e7d4bc153aa41c939bab3, 2013-05-13, lsblk: add SERIAL column
99f43b721f3dd31dcc8bdb1835c66c5b735aa63f, 2012-11-02, lsblk: Add vendor column
5fb4d23ebc8456869e548d7c899d57018abe571, 2012-10-19, lsblk: add SCSI transport type attribute
```
**After reviewing versions release dates, the minimum version of util-linux bootiso supports is 2.26**

So I should add so guard-code to prevent the issues you went into.
As per the partition issue reproduction, I'll work on it today, I pledge!

RonCam, 2018-08-21:
Thanks for pursuing the problem.  The check-box in Synaptic for util-linux_2.20.1-5 is solid green, meaning no updates to a more recent version of _bootiso_ are available, in the Bodhi/Ubuntu repo for Trusty.  Question: is there any way of forcing an updated version of util-linux to load, from a different source? 

Unless this should be possible, am I correct that the only fix for my system is applying your patches to the script?  And, then should the USB drive-detection then begin to work?  Or, is that a separate issue?

About the Invalid partition table - recursive partition error, as detected when opening the PNY drive with gparted: if you see the same problem, then perhaps it's not "buggy firmware" after all, but instead a problem built-into the tails .iso.  It will be interesting to see if you can duplicate that error, or not.

HassanMirza01, 2018-08-22:
parted is all in one solution to all stuff, it can do anything what gparted can do, inface, parted is cli version of gparted...

RonCam, 2018-08-22:
@jsamr @HassanMirza01 :  Past exprience with _gparted_ shows it to be extremely sensitive to file-structure abnormalities, to which other utilities, file managers, etc. would just 'give a pass'.

So, even if I could manage to put another partition on it by other means, if 'something' is causing the drive to be produced with a defective file structure, I would be reluctant to trust that partition -- or the drive itself, for that matter -- for any serious use.

By the way @jsamr , that work-around for adding a new partition is most interesting and thanks for bringing it to my attention!  I didn't know of that, before ...

RonCam, 2018-08-22:
@jsamr : On my PNY boot disk, _parted_ (not _gparted_, this time) is giving the same error, saying there's something wrong with the filesystem:
```
~$ sudo parted /dev/sdc
GNU Parted 2.3
Using /dev/sdc
Welcome to GNU Parted! Type 'help' to view a list of commands.
(parted) print                                                            
**Error: Invalid partition table - recursive partition on /dev/sdc.**         
Ignore/Cancel?    
```
However, the utility you used to open and manipulate the file system, _cfdisk_, did so without complaining.   

Any ideas on where the problem may lie?  Does the boot drive you created for tails open properly with _parted_?

jsamr, 2018-08-22:
@RonCam I could reproduce your issue with `parted`. 
I have no idea what is wrong here. But I suspect tails ISO to be somehow defective. `bootiso` only use `dd` on the background, which performs a byte-to-byte exact copy of the image.

RonCam, 2018-08-22:
@jsamr : Yes, I agree, this is the most likely explanation.  

Out of curiosity, I will now be looking for other utilities to examine specifically for "invalid partition tables" and "recursive partitions" to see if anything other than _parted_ or _gparted_ detects a problem.  

In any event, thanks for taking the time to pursue this and duplicating the problem on your system!  Now I know there would be no point in discarding the PNY drive, installing different kernels, different versions of `util-linux,` etc.,  in an attempt get parted/gparted to work with this flash drive, following the tails 'burn'.  

I'll continue using your script, with the patches you provided, whenever I need to 'burn' a flash drive with an iso.

RonCam, 2018-08-22:
I think I see the problem.  This is the tails flash boot disc created by `bootiso`, after running `sudo fdisk -l /dev/sdc` on it:
```
Disk /dev/sdc: 15.7 GB, 15728640000 bytes
255 heads, 63 sectors/track, 1912 cylinders, total 30720000 sectors
Units = sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disk identifier: 0x0000002a
   Device Boot      Start         End      Blocks   Id  System
/dev/sdc1   *           0     2377619     1188810   17  Hidden HPFS/NTFS
```
The first cylinder starts at 'zero', so therefore the MBR has been overwritten.  Does this seem correct or do you see something else?

If this is the problem, then is it something for which `bootiso` can compensate?

jsamr, 2018-08-23:
@RonCam I get sensibly the same output on both the image (iso) and the usb drive. The fact it starts at 0 is not an issue. I have tested a lot of ISOs, and they all start at 0 sector, which therefore **must** be relative to the end of MBR sector (512 bytes). When I see that an ISO has a partition table with file utility (test `file -b tails-iso...`), I consider it's "hybrid", and I presume the partition table is not broken. Bootiso just copy-paste and won't touch anything afterwards.

[image]

If you are patient, you could analyse byte by byte the MBR partition. To do so (here sdd is USB)

```
sudo dd if=/dev/sdd bs=512 count=1 2> /dev/null | hexdump -C
```
You should have the following output, in which you can see the MBR is not zeroed:

```
00000000  33 ed 90 90 90 90 90 90  90 90 90 90 90 90 90 90  |3...............|
00000010  90 90 90 90 90 90 90 90  90 90 90 90 90 90 90 90  |................|
00000020  33 ed fa 8e d5 bc 00 7c  fb fc 66 31 db 66 31 c9  |3......|..f1.f1.|
00000030  66 53 66 51 06 57 8e dd  8e c5 52 be 00 7c bf 00  |fSfQ.W....R..|..|
00000040  06 b9 00 01 f3 a5 ea 4b  06 00 00 52 b4 41 bb aa  |.......K...R.A..|
00000050  55 31 c9 30 f6 f9 cd 13  72 16 81 fb 55 aa 75 10  |U1.0....r...U.u.|
00000060  83 e1 01 74 0b 66 c7 06  f3 06 b4 42 eb 15 eb 02  |...t.f.....B....|
00000070  31 c9 5a 51 b4 08 cd 13  5b 0f b6 c6 40 50 83 e1  |1.ZQ....[...@P..|
00000080  3f 51 f7 e1 53 52 50 bb  00 7c b9 04 00 66 a1 b0  |?Q..SRP..|...f..|
00000090  07 e8 44 00 0f 82 80 00  66 40 80 c7 02 e2 f2 66  |..D.....f@.....f|
000000a0  81 3e 40 7c fb c0 78 70  75 09 fa bc ec 7b ea 44  |.>@|..xpu....{.D|
000000b0  7c 00 00 e8 83 00 69 73  6f 6c 69 6e 75 78 2e 62  ||.....isolinux.b|
000000c0  69 6e 20 6d 69 73 73 69  6e 67 20 6f 72 20 63 6f  |in missing or co|
000000d0  72 72 75 70 74 2e 0d 0a  66 60 66 31 d2 66 03 06  |rrupt...f`f1.f..|
000000e0  f8 7b 66 13 16 fc 7b 66  52 66 50 06 53 6a 01 6a  |.{f...{fRfP.Sj.j|
000000f0  10 89 e6 66 f7 36 e8 7b  c0 e4 06 88 e1 88 c5 92  |...f.6.{........|
00000100  f6 36 ee 7b 88 c6 08 e1  41 b8 01 02 8a 16 f2 7b  |.6.{....A......{|
00000110  cd 13 8d 64 10 66 61 c3  e8 1e 00 4f 70 65 72 61  |...d.fa....Opera|
00000120  74 69 6e 67 20 73 79 73  74 65 6d 20 6c 6f 61 64  |ting system load|
00000130  20 65 72 72 6f 72 2e 0d  0a 5e ac b4 0e 8a 3e 62  | error...^....>b|
00000140  04 b3 07 cd 10 3c 0a 75  f1 cd 18 f4 eb fd 00 00  |.....<.u........|
00000150  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001b0  40 01 00 00 00 00 00 00  2a 00 00 00 00 00 80 00  |@.......*.......|
000001c0  01 00 17 fe 3f 93 00 00  00 00 94 47 24 00 00 00  |....?......G$...|
000001d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
000001f0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 55 aa  |..............U.|
00000200
```

16 bytes of first partition definition are here:

```
                                                     ↓
000001b0  40 01 00 00 00 00 00 00  2a 00 00 00 00 00 80 00  |@.......*.......|
000001c0  01 00 17 fe 3f 93 00 00  00 00 94 47 24 00 00 00  |....?......G$...|
                                                   ↑
```

Now I won't go into analysing this! But you can if you wish. Here are some resources I found:
- [A cool guide](http://k5rec.blogspot.com/2006/11/linux-analyzing-master-boot-record-and.html)
- [Wikipedia entry with a detailed byte to byte description](https://en.wikipedia.org/wiki/Master_boot_record#Partition_table_entries)

RonCam, 2018-08-23:
Hi @jsamr ,
Thanks very much for the informative explanation.  I had not idea everything could be in the same cylinder ... : -?

I suppose I'm still puzzling over whatever made parted/gparted give that error message, and thought I had found an answer -- but apparently not.  Then, if it has a partition table, the other idea of using the free space for data storage may still be OK -- or, at least, it's not impossible.  

Now I can continue using the PNY drive without worrying it's contents are likely to disappear without warning ... 

Thanks for promising the next version, to allow detection and display of the USB parameters!  And, I'm glad if what I found when running the script on my older system is helping make _bootiso_ better.

closed 2018-08-23 by RonCam