jsamr/bootiso #29: Bootiso: Could not find a SYSLINUX bios folder containing c32 bios module files on this system.

c9ghtst, 2018-08-30:
Bootiso fails to run on Fedora 27 system with syslinux packages installed. 

[root@fedora27 ~]# ~user/bin/bootiso /local/win10/Win10_1803_English_x64.iso 
find: ‘/usr/lib/syslinux/’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
bootiso: Could not find a SYSLINUX bios folder containing c32 bios module files on this system.
         Exiting...

[root@fedora27 ~]# rpm -qa | grep syslinux
syslinux-6.04-0.7.fc27.x86_64
syslinux-extlinux-6.04-0.7.fc27.x86_64
syslinux-nonlinux-6.04-0.7.fc27.noarch
syslinux-extlinux-nonlinux-6.04-0.7.fc27.noarch

Script checks the presence of folder /usr/lib/syslinux/bios and file mbr.bin in bios sub-folder. Folder /usr/lib/syslinux doesn't exist in the file system. Package syslinux-nonlinux-6.04-0.7.fc27.noarch has *.c32 and *.bin files installed in /usr/shares/syslinux/
As a workaround folder /usr/lib/syslinux/bios can be created manually and synced with /usr/shares/syslinux/

[root@fedora27 ~]# rpm -ql syslinux-nonlinux
/usr/share/syslinux/altmbr.bin
/usr/share/syslinux/altmbr_c.bin
/usr/share/syslinux/altmbr_f.bin
/usr/share/syslinux/cat.c32
/usr/share/syslinux/chain.c32
/usr/share/syslinux/cmd.c32
...
[root@fedora27 ~]# find / -type f -name "mbr.bin"
/usr/share/syslinux/mbr.bin
[root@fedora27 ~]# mkdir -p /usr/lib/syslinux/bios
[root@fedora27 ~]# rsync -av /usr/share/syslinux/ /usr/lib/syslinux/bios/
[root@fedora27 ~]# ~user/bin/bootiso /local/win10/Win10_1803_English_x64.iso 
bootiso: Found non-hybrid ISO; inspecting ISO for boot capabilities...
bootiso: UEFI boot check validated. Your USB will work with UEFI boot.
bootiso: Partition label automatically set to 'CCCOMA_X64F'.
         You can explicitly set label with '-L, --label' option.
bootiso: Listing drives available in your system:
         NAME                 MODEL            VENDOR     SIZE TRAN   HOTPLUG SERIAL
         sdf                       USB                 Flash           7.6G usb                     1 
bootiso: Autoselecting 'sdf' (only USB device candidate)
bootiso: The selected device '/dev/sdf' is connected through USB.
bootiso: Created ISO mount point at '/mnt/iso.0TL'.
bootiso: About to wipe out the content of device '/dev/sdf'.
         Are you sure you want to proceed? (y/n)>y
bootiso: Erasing contents of '/dev/sdf'
...
bootiso: Created USB device mount point at '/mnt/usb.1Wb'
bootiso: Copying files from ISO to USB device with 'rsync'    
bootiso: Synchronizing writes on device '/dev/sdf'    
bootiso: Took 870 seconds to perform [install-auto] action.
bootiso: ISO succesfully unmounted (/mnt/iso.0TL).
bootiso: USB device partition succesfully unmounted.
bootiso: USB device succesfully ejected.
         You can safely remove it!

jsamr, 2019-01-27:
@c9ghtst With v3.2.0, you can export `SYSLINUX_LIB_ROOT` environment variable to `/usr/share/syslinux`, which should fix the issue. Would be nice to confirm :-)

Ninj0r, 2019-02-10:
I can confirm that setting `SYSLINUX_LIB_ROOT` does not work with v3.2.2
```
echo $SYSLINUX_LIB_ROOT
/usr/share/syslinux

./bootiso ~/Downloads/Win10_1809Oct_English_x64.iso
bootiso: Found non-hybrid ISO; inspecting ISO for boot capabilities...
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
bootiso: Could not find a SYSLINUX bios folder containing c32 bios module files on this system.
         Exiting...
```

jsamr, 2019-02-10:
@Ninj0r Thanks for the feedback ; I'll take a look at it tomorrow.

jsamr, 2019-02-11:
@Ninj0r I have trouble finding an explanation to your outcome.

I added those lines after [L124](https://github.com/jsamr/bootiso/blob/master/bootiso#L124):

```
echo "$syslinuxLibRoot"
exit 0
```

And did

[image]

Are you absolutely certain the env variable is exported ? Can you try with an inline env set?

Ninj0r, 2019-02-12:
I removed the files at /usr/lib/syslinux/bios/ and tried it again after a reboot:
```
[Utils]$ uname -r
4.20.6-100.fc28.x86_64
[Utils]$ echo $SYSLINUX_LIB_ROOT

[Utils]$ 
```

So that variable isn't set; then:
```
[Utils]$ export SYSLINUX_LIB_ROOT=/usr/share/syslinux
[Utils]$ echo $SYSLINUX_LIB_ROOT
/usr/share/syslinux
[Utils]$ 
```

Which should mean the variable is set.
```
[Utils]$ ./bootiso ~/Downloads/Win10_1809Oct_English_x64.iso
bootiso: Found non-hybrid ISO; inspecting ISO for boot capabilities...
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
bootiso: Could not find a SYSLINUX bios folder containing c32 bios module files on this system.
         Exiting...
[Utils]$
```

Echoing inline:
```
[Utils]$ echo $SYSLINUX_LIB_ROOT; ./bootiso ~/Downloads/Win10_1809Oct_English_x64.iso
/usr/share/syslinux
bootiso: Found non-hybrid ISO; inspecting ISO for boot capabilities...
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
bootiso: Could not find a SYSLINUX bios folder containing c32 bios module files on this system.
         Exiting...
[Utils]$ 
```

So according to echo it should be set. Setting inline:
```
[Utils]$ export SYSLINUX_LIB_ROOT=/usr/share/syslinux; ./bootiso ~/Downloads/Win10_1809Oct_English_x64.iso
bootiso: Found non-hybrid ISO; inspecting ISO for boot capabilities...
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
bootiso: Could not find a SYSLINUX bios folder containing c32 bios module files on this system.
         Exiting...
[Utils]$ 
```

After modifying L124:
```
[Utils]$ ./bootiso 
/usr/share/syslinux
```

So it seems that `$syslinuxLibRoot` is getting the set value...! Very bizarre.

Looking a little closer:
```
L62 typeset syslinuxLibRoot=${SYSLINUX_LIB_ROOT:-'/usr/lib/syslinux'}
...
L1599   foundSyslinuxBiosFolder=$(find "$syslinuxLibRoot" -type d -path '*/bios' -print -quit)
L1600   foundSyslinuxMbrBinary=$(findFileFromPatterns "$syslinuxLibRoot" 'bios/mbr.bin' 'mbr.bin')
```
It's probably unable to find the files it needs and reverting to the default path.

If there's anything else I can do to help please let me know.

jsamr, 2019-02-12:
@Ninj0r Thank you very much for this detailed report.
I'll try to reproduce in a VM later this week and investigate further.

mutluerol, 2019-02-23:
i am using gentoo and tested for gentoo its working :+1:

soullivaneuh, 2019-03-17:
Same issue under void-linux, the env set does not work, have to do:

```
sudo mkdir -p /usr/lib/syslinux/bios
sudo rsync -av /usr/lib/syslinux/efi64 /usr/lib/syslinux/bios/
```

And then it works. The env var is ignored.

jadkik, 2019-03-31:
I got the same error and solved part of it by running the script with `sudo -E`. It seems at some point the script runs itself using `sudo` and the environment variables are lost in the process. The issue is still there but the error messages are different:

With `sudo -E`:

```
$ sudo -E ./bootiso ~/Downloads/Win10_Edu_1803_English_x64.iso 
potato /usr/share/syslinux
bootiso: Found non-hybrid ISO; inspecting ISO for boot capabilities...
bootiso: Could not find a SYSLINUX bios folder containing c32 bios module files on this system.
         Exiting...
```

Without `sudo -E`:

```
$ ./bootiso ~/Downloads/Win10_Edu_1803_English_x64.iso 
potato /usr/share/syslinux
potato /usr/lib/syslinux
bootiso: Found non-hybrid ISO; inspecting ISO for boot capabilities...
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
find: ‘/usr/lib/syslinux’: No such file or directory
bootiso: Could not find a SYSLINUX bios folder containing c32 bios module files on this system.
         Exiting...
```

Then I tried making a symlink (`ln -sv /usr/share/syslinux ./syslinux/bios`) and it didn't pick it up because internally in the `find` command it's looking for a directory.

This is what ended up working for me:

```
mkdir -p syslinux/bios/
rsync -av /usr/share/syslinux/ syslinux/bios/
export SYSLINUX_LIB_ROOT="`pwd`/syslinux"
sudo --preserve-env=SYSLINUX_LIB_ROOT ./bootiso ~/Downloads/Win10_Edu_1803_English_x64.iso
```

jsamr, 2019-04-01:
@jadkik Thanks a lot for pointing the `sudo -E` and lost environment workout. I'll work on this soon.

closed 2019-04-04 by jsamr

jsamr, 2019-04-04:
Should be fixed in v3.3.0, let me know!

jadkik, 2019-04-04:
I just tried it with `export SYSLINUX_LIB_ROOT=/usr/share/syslinux` and it still fails, but for another reason. It's because of the directory structure in `/usr/share/syslinux` is different (there is no `bios` directory).

Here's the structure that is found on Fedora 29:

```
$ find $SYSLINUX_LIB_ROOT -type d
/usr/share/syslinux
/usr/share/syslinux/efi64
/usr/share/syslinux/dosutil
/usr/share/syslinux/diag
/usr/share/syslinux/com32
/usr/share/syslinux/com32/include
/usr/share/syslinux/com32/include/syslinux
/usr/share/syslinux/com32/include/linux
/usr/share/syslinux/com32/include/sys
/usr/share/syslinux/com32/include/sys/i386
/usr/share/syslinux/com32/include/sys/x86_64
/usr/share/syslinux/com32/include/gplinclude
/usr/share/syslinux/com32/include/gplinclude/zzjson
/usr/share/syslinux/com32/include/gplinclude/acpi
/usr/share/syslinux/com32/include/gplinclude/disk
/usr/share/syslinux/com32/include/gplinclude/dmi
/usr/share/syslinux/com32/include/gplinclude/vpd
/usr/share/syslinux/com32/include/netinet
/usr/share/syslinux/com32/include/bitsize32
/usr/share/syslinux/com32/include/bitsize
/usr/share/syslinux/com32/include/hw
/usr/share/syslinux/com32/include/klibc
/usr/share/syslinux/com32/include/klibc/i386
/usr/share/syslinux/com32/include/klibc/x86_64
/usr/share/syslinux/com32/include/bitsize64
```

And the one that actually works:

```
find ./syslinux/ -type d
./syslinux/
./syslinux/bios
./syslinux/bios/efi64
./syslinux/bios/dosutil
./syslinux/bios/diag
./syslinux/bios/com32
./syslinux/bios/com32/include
./syslinux/bios/com32/include/hw
./syslinux/bios/com32/include/bitsize32
./syslinux/bios/com32/include/netinet
./syslinux/bios/com32/include/syslinux
./syslinux/bios/com32/include/linux
./syslinux/bios/com32/include/sys
./syslinux/bios/com32/include/sys/x86_64
./syslinux/bios/com32/include/sys/i386
./syslinux/bios/com32/include/bitsize
./syslinux/bios/com32/include/klibc
./syslinux/bios/com32/include/klibc/x86_64
./syslinux/bios/com32/include/klibc/i386
./syslinux/bios/com32/include/gplinclude
./syslinux/bios/com32/include/gplinclude/vpd
./syslinux/bios/com32/include/gplinclude/disk
./syslinux/bios/com32/include/gplinclude/acpi
./syslinux/bios/com32/include/gplinclude/zzjson
./syslinux/bios/com32/include/gplinclude/dmi
./syslinux/bios/com32/include/bitsize64
```

I guess you could mention in the README the `rsync` commands to get the correct files in the structure you expect, or maybe look for the files in either `$SYSLINUX_LIB_ROOT/bios/` or `$SYSLINUX_LIB_ROOT/`.

jsamr, 2019-04-04:
@jadkik Have you tried with `export SYSLINUX_LIB_ROOT=/usr/share/syslinux/bios` ?

jadkik, 2019-04-04:
There is no such directory by default. That's my setup:

```
$ rpm -qa | grep syslinux
syslinux-extlinux-nonlinux-6.04-0.8.fc28.noarch
syslinux-6.04-0.8.fc28.x86_64
syslinux-tftpboot-6.04-0.8.fc28.noarch
syslinux-extlinux-6.04-0.8.fc28.x86_64
syslinux-perl-6.04-0.8.fc28.x86_64
syslinux-efi64-6.04-0.8.fc28.x86_64
syslinux-nonlinux-6.04-0.8.fc28.noarch
syslinux-devel-6.04-0.8.fc28.x86_64
$ cat /etc/os-release 
NAME=Fedora
VERSION="29 (Twenty Nine)"
ID=fedora
VERSION_ID=29
(...)
```

Civil, 2019-09-01:
Same on Gentoo with syslinux-6.04.

It seems that it would be great if bootiso will automatically check /usr/share/syslinux and, for example, check for `bios` directory and if it's not there - for one of the files it expects and based on results it'll set corresponding variables.

Otherwise workaround is to do something like:
```
mkdir -p ~/syslinux/bios
cp -a /usr/share/syslinux/* ~/syslinux/bios/
export SYSLINUX_LIB_ROOT="~/syslinux"
bootiso PATH_TO_ISO
rm -rf ~/syslinux
```

commit 438eb3b6 (2019-04-04): Fix #29, add --preserve-env flag to sudo invocation