jsamr/bootiso #31: add support for using fdisk if sfdisk is not present and user selects n

prototype99, 2018-11-11:
reason being: in gentoo util-linux does not contain sfdisk and there is no ebuild for it (low demand? who knows) so there is no easy way to resolve that. however, my installation anyway and i imagine most others as well, does contain fdisk. being fairly universal (from my experience) i imagine this would resolve other potential edge case scenarios as well.

jsamr, 2018-12-15:
@prototype99 Do you know if `sfdisk` is available as a separate package in gentoo ? `sfdisk` is specialized for script usage so I am reluctant to fallback to `fdisk`, with the possible introduction of bugs.

prototype99, 2018-12-16:
https://gpo.zugaina.org/Search?search=sfdisk as you can see returns nothing (the site searches both official and unofficial packages) and https://gpo.zugaina.org/sys-apps/util-linux is what provides fdisk but no sfdisk unfortunately or at least that was the case. strangely enough sfdisk is now here, whether someone saw this and fixed util-linux without saying anything or what i don't know. previously the script said there was no sfdisk and if i tried to run sfdisk the command wasn't recognised so it clearly wasn't installed.

jsamr, 2019-01-05:
@prototype99 Thanks! I am closing this issue. Please reopen if necessary.

closed 2019-01-05 by jsamr