jsamr/bootiso #32: Support splitting of install.wim via "wimlib-imagex split" to avoid hitting 4GB file size limit

core-process, 2018-12-08:
https://wimlib.net/ support splitting of wim files. Exclude `install.wim` from rsync process and copy via the `split` command of `wimlib-imagex` instead.

core-process, 2018-12-15:
@jsamr Please have a look at https://github.com/core-process/wininst-uefi-usb/blob/master/README.md. It is macOS specific, but the copy sequence is identical to the Linux variant.

Most important part is:
```sh
# Copy all files except install.wim
rsync -v -r -I --no-links --no-perms --no-owner --no-group --exclude sources/install.wim /Volumes/WINISO/ /Volumes/WININST/

# Split-copy install.wim
wimlib-imagex split /Volumes/WINISO/sources/install.wim /Volumes/WININST/sources/install.swm 1024
```

jsamr, 2018-12-15:
@core-process I'm reluctant to add image-specific logic to a bash script ! This is mainly a maintainability concern.

However, I have been thinking of a `go` or `rust` successor to `bootiso` which could have more fine grained logic such as the example you are proposing :slightly_smiling_face:

core-process, 2018-12-15:
@jsamr I understand your concern. But this specific issue is very real and rendered the otherwise very beautiful bootiso script useless to me. :-( The Windows ISO images available at the official download site of Microsoft contains an install.wim file which is larger than 4 GB.

jsamr, 2018-12-16:
@core-process Oh I didn't catch the motivations at first glance. I will consider this option for next release. I'll ping you when that happens!

bboti86, 2018-12-21:
@jsamr have you considered using NTFS instead of vfat when the .wim file is too large for FAT? That could be an extra sanity check instead of processing the wim file.

core-process, 2018-12-22:
@bboti86 NTFS is not guaranteed to work, as it is not part of the standard. And I can confirm it does not work in many cases. FAT32 + processing the WIM file does work with any compliant system.

jsamr, 2019-01-27:
@core-process Currently implementing and testing ; should be shipped with 3.2 soon!

closed 2019-01-27 by jsamr

NullVoxPopuli, 2019-02-15:
Using the master branch, I just tried making a Win10 usb installer. The iso is over 5GB, so, WoeUSB doesn't seem to work for me, becaues of this error: 
[image]

NullVoxPopuli, 2019-02-15:
however, the CLI works
```bash
sudo woeusb --target-filesystem NTFS --device ~/Downloads/Win10_1809Oct_English_x64.iso /dev/sdf
```

jsamr, 2019-02-15:
@NullVoxPopuli This is not woeusb ; but bootiso :-) Guess you have been misguided by your favorite search engine.

NullVoxPopuli, 2019-02-15:
lol. oops

paulkinzo, 2021-01-07:
I found out wimlib-imagex not working on my Mac with latest Big Sur. The usb is not bootable. A possible solution is to download an old version of Windows 10 (before 2018 release) from [here](https://tb.rg-adguard.net/public.php) and the install.wim file will be less than 4GB so no need to split the image file. 

After that, burn Windows 10 to USB with [UUByte ISO Editor (Mac)](https://www.uubyte.com/iso-editor.html) or [Boot Camp Assistant](https://support.apple.com/en-us/HT201468). This procedure works flawlessly on Mac without splitting the ISO image.