jsamr/bootiso #43: Replace dot for dash in temporary folder names when using wimlib

Ill1, 2019-02-11:
Wimlib-imagex searches for the first dot and puts a number behind it to generate successive image splits.
The bootiso scripts generates mount points with dots in the folder names.
This causes the script to fail when using wimlib-imagex because it tries to write the second, third and fourth... files to the wrong temporary folder (mount points).

jsamr, 2019-02-11:
@Ill1 Thanks for your feedback.
I have successfully used and tested the wimsplit feature with wimlib 1.13.0, before releasing.

Can you give me your system information (neofetch) + wimlib version to help investigate this issue.

Also, it would be nice of you to use a title which express the issue you encountered instead of the suggested solution, so that I can mark it as `bug`.

Ill1, 2019-02-12:
@jsamr Sure:
wimlib-imagex (distributed with wimlib 1.12.0)
Copyright (C) 2012-2017 Eric Biggers

Kernel: 4.15.0-45-generic 
Shell: bash 4.4.19 
DE: GNOME 3.28.3 
WM: GNOME Shell 
Terminal: gnome-terminal 

Apologies for including the how-to-fix rather than what-was-broken - The fix worked for me on local, I initially tried to setup a Pull Request with the same.

jsamr, 2019-02-12:
@Ill1 OK. You are describing a wimlib bug, corrected in 1.13, which has a nasty side effect on bootiso!
[From their changelog:](https://wimlib.net/git/?p=wimlib;a=blob;f=NEWS;h=a727ebb1edfa79ec13761363157f85fdffb262a1;hb=HEAD#l1)

>  wimsplit (API: wimlib_split()) now correctly handles a dot in the path to the first split WIM part, prior to the filename extension.

So I'll publish a patch to have retro-compatibility on earlier versions of wimsplit.

Ill1, 2019-02-12:
Thanks @jsamr

jsamr, 2019-02-12:
@Ill1 I'll keep it open until released for others stumbling on this :-)

closed 2019-04-04 by jsamr

jsamr, 2019-04-04:
Should be fixed in v3.3.0, let me know!

commit 092df6ec (2019-04-04): Fix #43, remove dot from temporary files and folders templates