jsamr/bootiso #46: Failed to open "/mnt/usb2.BJM/sources/install.swm" for writing: No such file or directory

mdPlusPlus, 2019-03-21:
**Category**

- *behavior* (bootiso didn't behave as promised)

**Describe the bug**

Following output:

```
bootiso: Detected a Windows install.wim file, which will be handled by 'wimlib-imagex' utility.
bootiso: Copying files from ISO to USB device with 'rsync'   \
         Writing "/mnt/usb.BJM/sources/install.swm" (part 1 of 5): 0 MiB of 4170 MiB (0%) written
|         Writing "/mnt/usb2.BJM/sources/install.swm" (part 2 of 5): 1022 MiB of 4170 MiB (24%) written
[ERROR] Failed to open "/mnt/usb2.BJM/sources/install.swm" for writing: No such file or directory
         ERROR: Exiting with error code 47:
                Failed to open a file.
 
bootiso: Synchronizing writes on device '/dev/sde'    
bootiso: Took 23 seconds to perform [install-auto] action.
bootiso: ISO succesfully unmounted (/mnt/iso.Kcx).
```

Then hangs for a few minutes and outputs:

```
bootiso: USB device succesfully ejected.
         You can safely remove it!
```

During the hanging time so can see "umount /mnt/usb.XYZ" via `ps aux`. Unmount command can not be killed with `sudo kill -9 PID`. Bootiso can not be canceled with CTRL+C during that time.

**Expected behavior** (for behavioral bugs only)

Should not throw "No such file or directory" error and hang. Should exit successfully

**To Reproduce**

Steps to reproduce the behavior:

`./bootiso ./isos/Win10_1809Oct_German_x64.iso`

Error occurs every time, restarting the system does not help.

**Environment:**

```
OS: Ubuntu 18.10 x86_64 
Kernel: 5.0.0-acso 
Uptime: 12 mins 
Packages: 2860 (dpkg), 10 (snap) 
Shell: bash 4.4.19 
Resolution: 1920x1200, 1920x1080 
DE: GNOME 3.30.2 
WM: GNOME Shell 
WM Theme: Adwaita 
Theme: Yaru [GTK2/3] 
Icons: Yaru [GTK2/3] 
Terminal: gnome-terminal 
CPU: AMD Ryzen 7 1700X (16) @ 3.400GHz 
GPU: AMD ATI Radeon RX Vega 64 
GPU: AMD ATI Radeon RX 550/550X 
Memory: 1913MiB / 32159MiB
```

bootiso 3.2.2
wimlib-imagex 1.12.0
extlinux 6.04

**Additional context** 

8GB Verbatim USB stick

**Suggestion fix**

Currently no idea, will have to investigate.

jsamr, 2019-03-21:
@mdPlusPlus I believe it's related to #43 if your version of wimlib is bellow 1.13

mdPlusPlus, 2019-03-21:
Is there an official .deb or PPA for 1.13?

closed 2019-03-21 by mdPlusPlus

jsamr, 2019-04-04:
Should be fixed in v3.3.0, let me know!