jsamr/bootiso #48: Failure and inconsistency

Brottweiler, 2019-04-16:
**Category**

- *behavior* (bootiso didn't behave as promised)

**Describe the bug**

I hate using `dd`, and i dislike the GUIs (like etcher), so bootiso seemed to fit my case. When using it however, it seems like it succeeds, but actually using the USB stick to boot with causes failure.

I copied Ubuntu MATE 18.04.2 LTS with bootiso, and booted my old laptop with it. The Ubuntu MATE logo shows up with 5 dots. Two dots fill up, then freeze. Starting with `nomodeset` shows me errors scroll by, then getting a kernel panic.

I try another USB stick. Same issue, but this time the errors are different, then kernel panic.

I then try plain old `dd` and it worked like intended, I can boot the laptop with the USB stick and install MATE.

**Expected behavior** (for behavioral bugs only)

I expect bootiso to work as intended so I can boot the computer with my USB stick and install the distro.

**To Reproduce**

1. Run `bootiso ubuntu-mate-18.04.2-desktop-amd64.iso` with [this ISO](http://cdimage.ubuntu.com/ubuntu-mate/releases/18.04/release/ubuntu-mate-18.04.2-desktop-amd64.iso)

2. See error log:

I don't have the log for bootiso anymore (unless its logged to a file somewhere), but I can tell you the output was fine. Bootiso said the action was fine, and bootiso was using `dd` because it said it was using `dd`.

**Environment:**

```
OS: Arch Linux x86_64
Kernel: 5.0.7-arch1-1-ARCH
Packages: 1095 (pacman)
Shell: zsh
WM: Openbox
Terminal: st
Terminal Font: Misc Fixed
CPU: i5-4460 (4) @ 3.4GHz
GPU: GeForce GTX 1050 Ti
Memory: [━━━━━━━━━━━━━━━] 2492MiB / 7840MiB
Disk (/): [━━━━━━━━━━━━━━━] 29G / 110G (28%)
Disk (home): [━━━━━━━━━━━━━━━] 77G / 439G (19%)
```

**Additional context** (optional)

As I said, I was using `dd bs=4M if=path/to/archlinux.iso of=/dev/sdx status=progress oflag=sync` as adviced [here](https://wiki.archlinux.org/index.php/USB_flash_installation_media#Using_dd)] just fine, where bootiso failed, twice, on two sticks.

jsamr, 2019-04-17:
@Brottweiler I did try to reproduce and I could boot without issue.

[image]

Normally, a call to `sync` after the `dd` invocation guarantees that all the blocks are transferred before ejecting the device. The corruption you are referring to makes me think this sync phase might have been skipped, or the device ejected too early. Did you notice any difference in the duration of those two tasks, `dd` and `bootiso` ?

`oflag=sync` seems to achieve the same purpose.

I forgot to include bootiso version in the bug template ; I bet you are using the latest 3.3.1?

If so, you could try to apply this patch (adding the sync option) and tell me if you can reproduce the same bug:

```patch
--- a/bootiso
+++ b/bootiso
@@ -1222,7 +1222,7 @@ copyWithDD() {
                typeset statusFile=$(createTempFile "bootiso-status")
                temporaryAssets+=("$statusFile")
                (
-                       dd if="$selectedIsoFile" of="$selectedDevice" bs=4MB status=none
+                       dd if="$selectedIsoFile" of="$selectedDevice" bs=4MB status=none oflag=sync
                        echo "$?" >"$statusFile"
                ) &
                pid=$!
```

Brottweiler, 2019-04-17:
Both `dd` and `bootiso` took roughly the same amount of time. One thing I did notice was that when `bootiso` said it was finished, it takes a really long time until bootiso exits and I get my prompt back.

I seem to be using 3.3.0 from the AUR, I just flagged it out of date. 

Also, `oflag=sync` is probably new-ish, because the recommended procedure used to be to just call `&& sync` afterwards but now it seems to be "integrated" with `dd`.

I'll try the patch when I get the chance. Thanks!

jsamr, 2020-04-18:
I stumbled on this line in `coreutils sync` info page: 
```
Note also that depending on the system,
passing individual device nodes or files may have different sync
characteristics than using no arguments.  I.e., arguments passed to
fsync(2) may provide greater guarantees through write barriers,
than a global sync(2) used when no arguments are provided.
```
So I am tempted to explicitly invoke `sync` with a device node argument pointing to the USB device. Will do for next release.

commit 7c621597 (2020-04-29): fix one sync call which was run into background

That certainly could lead to corrupt data.

fix #48
closed 2020-05-17 by jsamr