jsamr/bootiso #50: wimlib-imagex not found

Rocket-Wolf, 2019-06-21:
sudo bootiso
bootiso: Command 'wimlib-imagex' not found! Should be in package 'wimlib'.
Attempt installation? (y/n)> y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package wimlib
bootiso: Installation of dependency 'wimlib-imagex' failed.
         Perhaps this dependency has a slightly different name on your distribution.
         Find it and install manually.
         Exiting...

jsamr, 2019-06-21:
@Rocket-Wolf, well, perhaps the message is self explicit enough ; what distribution are you using and which package manager ? Have you tried to find the exact name of wimlib program for this package manager?

Rocket-Wolf, 2019-06-21:
I followed the instructions for Quick install implicitly.
Using LM 19.1

Quick install
>
> curl -L https://git.io/bootiso -O
> chmod +x bootiso
>
> "If a man has to tell you he is a leader, he is not. If a woman has to
tell you she is a lady, she is not"  ~Author unknown

jsamr, 2019-06-21:
@Rocket-Wolf I just found out that on linux mint, the name of the package should be `wimtools`: https://community.linuxmint.com/software/view/wimtools

Please try `apt-get install wimtools` and try again to run bootiso.

KrzysztofHajdamowicz, 2019-06-25:
@jsamr this solution works for me. Can it be implemented in future release of bootiso?

jsamr, 2019-07-08:
@KrzysztofHajdamowicz  The selection of the appropriate package name per distro cannot exhaustively implemented. At some point, bootiso should be packaged by distro maintainers with the necessary set of dependencies.

closed 2019-07-08 by jsamr

espmartin, 2019-08-06:
I'm running Ubuntu 19.04, & got that same error. I installed wimtools from here:
https://launchpad.net/ubuntu/bionic/amd64/wimtools/1.12.0-1build1

adatum, 2019-08-13:
Same problem on Fedora 30. 

```
$ sudo ./bootiso 
bootiso: Command 'wimlib-imagex' not found! Should be in package 'wimlib'.
Attempt installation? (y/n)> y
No match for argument: wimlib
Error: Unable to find a match: wimlib
bootiso: Installation of dependency 'wimlib-imagex' failed.
         Perhaps this dependency has a slightly different name on your distribution.
         Find it and install manually.
         Exiting...

```

I can't find wimlib, wimlib-imagex, or wimtools in the repos.

```
$ sudo dnf provides *wimlib-imagex*
Error: No Matches found
```

```
$ sudo dnf search *wim*
======================== Name & Summary Matched: *wim* =========================
texlive-swimgraf-doc.noarch : Documentation for swimgraf
perl-WWW-Twilio-TwiML.noarch : Light and fast TwiML generator
texlive-swimgraf.noarch : Graphical/textual representations of swimming
                        : performances
============================= Name Matched: *wim* ==============================
perl-Swim.noarch : See What I Mean is a plain text markup language
============================ Summary Matched: *wim* ============================
perl-Check-ISA.noarch : DWIM, correct checking of an object's class
perl-MetaCPAN-API.noarch : A comprehensive, DWIM-featured API to MetaCPAN
perl-MetaCPAN-Client.noarch : A comprehensive, DWIM-featured client to the
                            : MetaCPAN API

```

jsamr, 2019-08-13:
@adatum Apparently, there doesn't seem to be an official port for Fedora, but you can install manually, see https://wimlib.net/forums/viewtopic.php?t=429&p=1041

adatum, 2019-10-17:
`wimtools` seems to be the package name in Debian-based distros.

I just checked again, and Fedora's repos now have `wimlib`:

```
$ dnf -C search wimlib
========================= Name Exactly Matched: wimlib =========================
wimlib.i686 : Open source Windows Imaging (WIM) library
wimlib.x86_64 : Open source Windows Imaging (WIM) library
======================== Name & Summary Matched: wimlib ========================
wimlib-devel.i686 : Development files for wimlib
wimlib-devel.x86_64 : Development files for wimlib
============================= Name Matched: wimlib =============================
wimlib-utils.x86_64 : Tools for creating, modifying, extracting, and mounting
                    : WIM files

```

```
$ dnf -C provides wimlib-imagex
wimlib-utils-1.13.1-1.fc30.x86_64 : Tools for creating, modifying, extracting,
                                  : and mounting WIM files
Repo        : updates
Matched from:
Filename    : /usr/bin/wimlib-imagex

```

So the package to install on Fedora is `wimlib-utils`, which also installs `wimlib` as a dependency.

Twaintes, 2020-01-14:
> @Rocket-Wolf I just found out that on linux mint, the name of the package should be `wimtools`: https://community.linuxmint.com/software/view/wimtools
> 
> Please try `apt-get install wimtools` and try again to run bootiso.

Worked for me, big thanks :)

harshgp, 2020-08-17:
> @Rocket-Wolf I just found out that on linux mint and debian-based systems, the name of the package should be `wimtools`: https://community.linuxmint.com/software/view/wimtools
> 
> Please try `apt-get install wimtools` and try again to run bootiso.
> 
> **EDIT**: for Fedora, use `wimlib-utils` (thanks @adatum)

worked :)

ibrahimbulut123, 2023-02-21:
i wrote this code but it didn't work. 
i clicked this link and i couldn't download. Please help me!