jsamr/bootiso #55: Feature request: Automatic handling of compressed images

emk2203, 2019-10-19:
It would be nice to have Etcher-like functionality to detect and handle compressed images like img.xz.

This in addition to handling img files in general.

jsamr, 2020-04-25:
@emk2203 The thing is, it would add a lot of new dependencies for what seems like an edge-case. What is your typical use case?

emk2203, 2020-04-25:
The typical use case would be just that: Taking a downloaded system image in `img.xz` format and having bootiso handle it transparently. These are used a lot for Raspberry Pi and other single-board computers. When using `dd`, I would just pipe the file from the decompressor, but with `bootiso`, it's not so easy.

The process would be:

* feed file into bootiso when prompted
   * bootiso uses as-is if img or iso file
   * with .xz extension, file gets passed through `xz`first for on-the-fly decompression

davidebeatrici, 2022-08-03:
For reference:

`xzcat file.img.xz | dd of=/dev/sdx`

jsamr, 2022-08-04:
> For reference:
> 
> `xzcat file.img.xz | dd of=/dev/sdx`

We don't use dd 100% of the time. For instance, to handle windows images or when we manually install a bootloader.

davidebeatrici, 2022-08-04:
Right. I believe supporting `xz` archives for Linux images would already be a great start though.