jsamr/bootiso #57: Dependencies

SibrenVasse, 2020-05-17:
In the new dependency overview:
> The corresponding packages in Arch Linux are mke2fs, dosfstools and ntfs-3g. Mandated dependencies are listed bellow.
The package mke2fs does not exist. Do you mean e2fsprogs?

> partprobe | parted
I thought this one was no longer required?

jsamr, 2020-05-17:
@SibrenVasse You are right on both, fixed.

closed 2020-05-17 by jsamr

jsamr, 2020-05-17:
@SibrenVasse Thanks for the AUR package upgrade!
I think you missed the `jq` dependency requirement.

Also, even though those packages are often shipped, would be very nice to explicitly depend on every package listed in the dependencies table!

SibrenVasse, 2020-05-17:
Ah forgot to re-add jq after fixing e2fsprogs!

As for explicitly depending on all packages (even if they are in 'base', so always installed):
I was under the assumption that was also not allowed for 'runtime dependencies', but it's only not allowed for makedepends and 'base-devel'.

jsamr, 2020-05-17:
@SibrenVasse You know better ;-)