jsamr/bootiso #60: FreeBSD

dariotarantini, 2020-06-18:
**Upstream URL:**

https://www.freebsd.org/
**Inspection:**

``` text
bootiso: Found non-hybrid image; inspecting image for boot capabilities... 
bootiso: The selected image is not hybrid, doesn't support UEFI or legacy booting 
         with SYSLINUX. Therefore, it cannot result in any successful booting with 
         bootiso. You can open an 'Image Support' request here: 
         https://github.com/jsamr/bootiso/issues/new/choose In the meantime, 
         consider following the instructions provided with this image file. 
         Exiting... 
```

jsamr, 2020-06-18:
@dariotarantini FreeBSD provides many different files depending on target architectures. For the sake of reproducibility, can you direct me to exactly which image file you used? Best would be a URL

jsamr, 2020-06-18:
@dariotarantini  With the memstick file, it works great:

[image]

jsamr, 2020-06-19:
I am closing now. Feel free to comment if you are still facing issues with FreeBSD. But as stated, it should work if you pick the appropriate `mimstick.img` file.

closed 2020-06-19 by jsamr