jsamr/bootiso #66: what should i do with "The SYSLINUX binary from kernel.org is for 32 bits architectures, but your system is 64 bits."

Xphobia, 2021-03-16:
**bootiso**: 4.2.0
 

 
**Category**: distro-dependant (when image SYSLINUX version doesn't match local version. )

**Describe the bug**
The local syslinux version is 4.05, while it's 6.03 in image Kylin-4.0.2-server-sp1-Community-20170412-x86_64.iso.
Then, bootiso will download syslinux 6.03 from website kernel.org. But the log shows 
```
"The SYSLINUX binary from kernel.org is for 32 bits architectures, but your system is 64 bits. If the install fails, you might need to install 32 bits support on your system."
```
What it means by "install 32 bits support on your system"?
What should i do to install 6.03 syslinux, not fail back to install local syslinux 4.05?

**Expected behavior** (for behavioral bugs only)
Install the right syslinux version 6.03

**Reproduction**
Run bootiso command as follws:
```
bootiso  --no-usb-check -J -y -d /dev/sdb Kylin-4.0.2-server-sp1-Community-20170412-x86_64.iso
```

**Exit status**
0

**Environment:**
1、OS distribution
```
[root@iso-test-13 images]# cat /etc/redhat-release 
CentOS Linux release 7.5.1804 (Core) 
```

2、Test image
```
Kylin-4.0.2-server-sp1-Community-20170412-x86_64.iso
```

3、Test log
```
[root@iso-test-13 images]# bootiso  --no-usb-check -J -y -d /dev/sdb Kylin-4.0.2-server-sp1-Community-20170412-x86_64.iso 
sfdisk: failed to dump partition table: Success
bootiso: Found non-hybrid image; inspecting image for boot capabilities... 
bootiso: Found SYSLINUX config file and binary at version 6.03.
bootiso: A SYSLINUX booloader will be installed on your USB device.
bootiso: Partition label automatically set to 'KYLIN-4.0.2'. You can explicitly set the label with -L, --label.
bootiso: USB check has been disabled. Skipping.
bootiso: About to wipe the content of device '/dev/sdb'.
bootiso: Bypassing confirmation with -y, --assume-yes.
bootiso: Erasing contents of '/dev/sdb'... 
bootiso: Creating MBR partition table with 'sfdisk' v2.29...
         Checking that no-one is using this disk right now ... OK 
         
         Disk /dev/sdb: 8 GiB, 8589934592 bytes, 16777216 sectors 
         Units: sectors of 1 * 512 = 512 bytes 
         Sector size (logical/physical): 512 bytes / 512 bytes 
         I/O size (minimum/optimal): 4096 bytes / 4194304 bytes 
         
         >>> Script header accepted. 
         >>> Created a new DOS disklabel with disk identifier 0xf39c822a. 
         /dev/sdb1: Created a new partition 1 of type 'W95 FAT32 (LBA)' and of 
         size 8 GiB. 
         Partition #1 contains a vfat signature. 
         The signature will be removed by a write command. 
         /dev/sdb2: Done. 
         
         New situation: 
         
         Device Boot Start End Sectors Size Id Type 
         /dev/sdb1 * 2048 16777215 16775168 8G c W95 FAT32 (LBA) 
         
         The partition table has been altered. 
         Calling ioctl() to re-read partition table. 
         Syncing disks. 
bootiso: Creating vfat partition on '/dev/sdb1'...
         mkfs.fat 3.0.20 (12 Jun 2013) 
         /dev/sdb1 has 64 heads and 32 sectors per track, 
         logical sector size is 512, 
         using 0xf8 media descriptor, with 16775168 sectors; 
         filesystem has 2 32-bit FATs and 8 sectors per cluster. 
         FAT size is 16351 sectors, and provides 2092804 clusters. 
         There are 32 reserved sectors. 
         Volume ID is 39b3c1cd, volume label KYLIN-4.0.2. 
bootiso: Created USB device mount point at '/var/tmp/bootiso/mnt/usb-JUU' 
bootiso: Copying files from image to USB device with 'rsync'    
bootiso: Synchronizing writes on device '/dev/sdb'   
bootiso: Found ISOLINUX config file at 
         '/var/tmp/bootiso/mnt/usb-JUU/isolinux/isolinux.cfg'. Moving to 
         '/var/tmp/bootiso/mnt/usb-JUU/isolinux/syslinux.cfg'. 
bootiso: Found local SYSLINUX version '4.05' 
bootiso: image SYSLINUX version doesn't match local version. Scheduling download of version 6.03...
bootiso: Found '/var/cache/bootiso/syslinux-6.03-pre9.tar.gz' in cache.
bootiso: SYSLINUX version '6.03' temporarily set for installation.
bootiso: Installing SYSLINUX bootloader in 
         '/var/tmp/bootiso/mnt/usb-JUU/isolinux' with kernel.org version 
         '6.03'... 
bootiso: The SYSLINUX binary from kernel.org is for 32 bits architectures, but your system is 64 bits. If the install fails, you might need to install 32 bits support on your system.
         /usr/local/bin/bootiso: 
         /var/tmp/bootiso/syslinux-6.03-pre9/bios/extlinux/extlinux: 
         /lib/ld-linux.so.2: bad ELF interpreter: No such file or directory 
bootiso: Could not run SYSLINUX '6.03' from kernel.org. Attempting with local SYSLINUX install...
bootiso: Successfully installed SYSLINUX bootloader at version '4.05'.
bootiso: Successfully installed Master Boot Record program.
bootiso: Took 135 seconds to perform install-auto action.
bootiso: USB device partition succesfully unmounted.
bootiso: USB device ejection skipped with -J, --no-eject. 

[root@iso-test-13 images]# echo $?
0

```

**Additional context** (optional)

**Suggestion fix** (optional)

jsamr, 2021-03-16:
@Xphobia You probably need glibc.i686, see https://stackoverflow.com/questions/14030306/lib-ld-linux-so-2-bad-elf-interpreter-no-such-file-or-directory

Xphobia, 2021-03-16:
> @Xphobia You probably need glibc.i686, see https://stackoverflow.com/questions/14030306/lib-ld-linux-so-2-bad-elf-interpreter-no-such-file-or-directory

Yes, thanks. After "yum install glibc.i686" , the problem solved.

closed 2021-03-16 by jsamr