jsamr/bootiso #69: fetches code online by default

jonassmedegaard, 2021-05-31:
bootiso by deault fetches SYSLINUX online if major version of local version is different than the one used in an image.

I find it it a nice feature that bootiso can fetch an alternate bootloader - but find it problematic that it is done by default.

Please consider flipping around the logic to operate offline by default.

jsamr, 2021-05-31:
@jonassmedegaard The rationale is that mismatching the major version leads to an important risk of producing an unbootable device (I remember trying that scenario with some popular recoveries ISOs). So I guess the handling of this scenario is very opinionated... What would be the rationale in support of the behavior you are advocating?

jonassmedegaard, 2021-05-31:
the rationale is that reading the man page description of `create a bootable USB drive from an image file` gives the impression that this is a tool moving IO from an image to a device.
To some, fetching executable code online and injecting that into the target device is no big deal, but to others it really is.
Like, what validation was made about the authenticity of that online code (was the PGP signature for the code itself validated against PGP signatures of authors of the code? Was the TLS certificate of the web hosting provider distributing the code validated against lists of revocation certificates and/or Google/Mozilla CA certificate blacklists?).
Or, was the user warned ahead that now their image flashing activity will be revealed to at least their own ISP and the ISP of the code blob distributor and the code blob distributor itself, but potentially also to the NSA via backbone routing providers?

jsamr, 2021-05-31:
@jonassmedegaard I understand that security is a good weight in the balance. I guess we could prompt the user to chose between the two options, unless he provides a flag explicitly preferring one option or the other. Would that address your concerns?

jonassmedegaard, 2021-05-31:
If you mean by default do nothing, only prompt, then I disagree.

I suggest to flip the logic from...
 * always flash, possibly with inferior code (and then emit a warning but continue without prompting) if network fails

...to...

 * always flash, possibly with inferior code (and then emit a warning but continue without prompting) if user has not explicitly permitted to fetch code online

jonassmedegaard, 2021-05-31:
...but yes it would address my concerns - just raise a different (not concern but) annoyance about suboptimal user experience :-)

jonassmedegaard, 2021-05-31:
I am thinking of something like this (untested!):
```diff
--- a/bootiso
+++ b/bootiso
@@ -1886,7 +1886,7 @@
   function _checkSyslinuxVersion() {
     local -i _versionsMatch=0
     if [ "$enableForceLocalBootloader" == true ]; then
-      term_echogood "Enforced local SYSLINUX bootloader at version '$_localSyslinuxVersion'."
+      term_echogood "Enforced local SYSLINUX bootloader at version '$_localSyslinuxVersion'.  Use $(term_boldify '--local-bootloader=false') to permit trying to download and execute a version from kernel.org."
       return 1
     fi
     if [ -z "$st_targetSyslinuxVersion" ]; then
@@ -2337,7 +2337,7 @@
   sourceHashFile=${st_userVars['hash-file']:-''}
   targetDevice=${st_userVars[device]:-''}
   targetPartitionLabel=${st_userVars[label]:-''}
-  targetBootloaderVersion=${st_userVars['remote-bootloader']:-'auto'}
+  targetBootloaderVersion=${st_userVars['remote-bootloader']:-''}
   targetDDBusSize=${st_userVars['dd-bs']:-'4M'}
   targetDataPartFstype=${st_userVars['data-part-fs']:-'vfat'}
   assumeImageIs=${st_userVars['assume-image-is']:-''}
@@ -2542,8 +2542,8 @@
       term_echoinfo "However, SYSLINUX version (${st_isoInspections[syslinuxVer]}) in the image file doesn't match the minor part of local version ($_localSyslinuxVersion), which should not cause any problems."
     else
       term_echowarn "SYSLINUX version (${st_isoInspections[syslinuxVer]}) in the image file doesn't match the major part of local version ($_localSyslinuxVersion)." \
-        "$scriptName will try to download and execute this version from kernel.org, unless given the modifier $(term_boldify '--local-bootloader')." \
-        "If that fails, it will attempt installation with the local version of SYSLINUX."
+        "$scriptName will attempt installation with the local version of SYSLINUX regardless."
+        "If that fails, you may try have it download and execute the exact version from kernel.org, by providing the modifier $(term_boldify '--remote-bootloader')."
     fi
   fi
 }
```

jsamr, 2021-05-31:
@jonassmedegaard That seems all right, I don't have strong feelings about either solution; would you be willing to submit a PR?

jonassmedegaard, 2021-05-31:
Sorry, I won't provide a PR: Github has a problematic terms of service with a term arguably equivalent to auto-granting a permissive license for all published code (i.e. defeating copyleft licensing like GPL).

What I can provide is a publicly accessible patch like the one quoted above - which (if you don't find time to do a better job at that than me) I will do as part of my Debian packaging.
I'd prefer that you do it, however, since you know the codebase better.

jsamr, 2021-05-31:
@jonassmedegaard I guess you're [referring to this](https://www.fsf.org/blogs/licensing/do-githubs-updated-terms-of-service-conflict-with-copyleft). But anyway, I can craft the patch (and yes, I know `git apply`); I'll also need to update the man and completions. Can't commit to a schedule, but I'll address this ASAP, along with other issues you've raised.

jonassmedegaard, 2021-06-03:
I am referring to same issue that FSF covers as well, but I recommend [this alternate view on the matter](https://joeyh.name/blog/entry/removing_everything_from_github/).

jsamr, 2021-06-29:
@jonassmedegaard It looks like the licensing issues aren't theoretical anymore with the new AI-powered GIthub Copilot, see https://news.ycombinator.com/item?id=27676266

In this hackernews thread, a team developer has been very evasive on the question of whether the Github AI could train on GPL-licensed work and generate suggestions based on copyleft licenses, thus seemingly infringing those licenses while passing under the radar. Indeed, an AI is a black box and there is no way to track down the myriad of original sources that have contributed to this suggestion. That is a morally challenging situation and I guess I have to at least consider migrating this repository to a different hosting platform.

Looks like there will be interesting legal fights ahead!